autowarefoundation/autoware #992: Calibration Toolkit not Receiving Point Cloud

ben-swain, 2017-12-19:
Point cloud is not received by calibration toolkit using `Camera -> Velodyne`.
The toolkit is not subscribing to the pointcloud topic named `/velodyne_points`.

#### Steps to Reproduce the Problem

  1. Play rosbag containing `/camera/image_raw` and `/velodyne_points`
  2. Launch Calibration Toolkit from Sensing tab
  3. Select image topic and `Camera -> Velodyne`

#### Expected Behavior
I expect the toolkit to subscribe to the `/velodyne_points` topic and display the point on the right panels.

#### Debug Info

[image]

@alexanderhmw mentioned in [this topic](https://github.com/CPFL/Autoware/issues/498) that since it says "timestamp," the toolkit is not receiving the point cloud, but I'm unsure how to fix this.

alexanderhmw, 2017-12-19:
The topic of received Velodyne points should be "/points_raw".
See:
https://github.com/CPFL/Autoware/blob/a5c51623d9055b28a33b8bdfc853a0dd1f5b6187/ros/src/sensing/fusion/packages/calibration_camera_lidar/nodes/calibration_toolkit/mainwindow.cpp#L87-L88

dejanpan, 2017-12-25:
@ben-swain it seems you got the fix, correct? if not reopen.

closed 2017-12-25 by dejanpan

ben-swain, 2017-12-25:
The issue was solved. Thanks @alexanderhmw