autowarefoundation/autoware #994: Cannot use GPU by launching dpm_ttic detector in PX2

MadCauchy, 2017-12-21:
Hallo Autoware:

I am trying to use the dpm-ttic detector in PX2. At first i set Use GPU as true, the detector doesn't work. The node is not launched. If i set it as false, the function can be started, but it performs not really fast. Because part of the advantage of PX2 is its powerful GPUs. I really hope it can be fixed.
the error message is as following:

> ========
> 
> PARAMETERS
> * /obj_car/dpm/comp_model_path: /home/nvidia/Auto...
> * /obj_car/dpm/detection_class_name: car
> * /obj_car/dpm/image_raw_topic: /camera/0/0/image...
> * /obj_car/dpm/part_model_path: /home/nvidia/Auto...
> * /obj_car/dpm/root_model_path: /home/nvidia/Auto...
> * /obj_car/dpm/use_gpu: True
> * /obj_person/dpm/comp_model_path: /home/nvidia/Auto...
> * /obj_person/dpm/detection_class_name: person
> * /obj_person/dpm/image_raw_topic: /camera/0/0/image...
> * /obj_person/dpm/part_model_path: /home/nvidia/Auto...
> * /obj_person/dpm/root_model_path: /home/nvidia/Auto...
> * /obj_person/dpm/use_gpu: True
> * /rosdistro: kinetic
> * /rosversion: 1.12.7
> 
> NODES
> /obj_car/
> dpm (cv_tracker/dpm_ttic)
> /obj_person/
> dpm (cv_tracker/dpm_ttic)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> [obj_person/dpm-2] process has died [pid 21212, exit code 1, cmd /home/nvidia/Autoware/ros/devel/lib/cv_tracker/dpm_ttic /config/obj_person/dpm:=/config/pedestrian_dpm __name:=dpm __log:=/home/nvidia/.ros/log/e78ca4ae-e5a5-11e7-b4b3-00044b707580/obj_person-dpm-2.log].
> log file: /home/nvidia/.ros/log/e78ca4ae-e5a5-11e7-b4b3-00044b707580/obj_person-dpm-2*.log
> [obj_car/dpm-1] process has died [pid 21211, exit code 1, cmd /home/nvidia/Autoware/ros/devel/lib/cv_tracker/dpm_ttic /config/obj_car/dpm:=/config/car_dpm __name:=dpm __log:=/home/nvidia/.ros/log/e78ca4ae-e5a5-11e7-b4b3-00044b707580/obj_car-dpm-1.log].
> log file: /home/nvidia/.ros/log/e78ca4ae-e5a5-11e7-b4b3-00044b707580/obj_car-dpm-1*.log
> process[obj_car/dpm-1]: started with pid [21211]
> process[obj_person/dpm-2]: started with pid [21212]
> all processes on machine have died, roslaunch will exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

and then i checked log file:

> car detection program 2 GPUs found
> car detection use GPU[No.0]
> car detection use GPU[No.1]
> 
> cuModuleload failed: res = CUDA_ERROR_INVALID_SOURCE

## Specifications

  - Ubuntu version:16.04
  - ROS Version:kinetic
  - Autoware branch: I followed the install instruction for PX2 provided by your group. 

Thank a lot!

yk-fujii, 2017-12-21:
We will deprecate dpm in ner future.
So, we recommend using DL based detector such as SSD or YOLO.(autoware can use these dl based detector)

Did you have to use dpm?
If so, please tell me the version of SDK on DPX2 you are currently using.

MadCauchy, 2017-12-21:
Thank you for your answer. I use the run file DriveInstall_4.1.8.0L_SDK_b43.run.  The version of SDK is 4.1.8.0. I will take your advise and try SSD or YOLO later.

dejanpan, 2017-12-25:
closing since @MadCauchy will use different detector.

closed 2017-12-25 by dejanpan

LUUTHIENXUAN, 2018-04-04:
I have the sample issue when launching dpm_ttic (with the same environment Ubuntu version:16.04 ROS Version:kinetic except 1 GPU found). For many reasons, I have to stick rigidly to "dpm_ttic", so any ideas how to fix it?