autowarefoundation/autoware #1000: Cannot Load Message: autoware_msgs/projection_matrix

ben-swain, 2017-12-22:
When I run Calibration Publisher in the Sensing tab and then `rostopic list`, I can see `/camera/projection_matrix` in the list, but `rostopic echo /camera/projection_matrix` results in the following error:

`ERROR: Cannot load message class for [autoware_msgs/projection_matrix]. Are your messages built?`

Overall, I'm trying to use Points Image, but it requires `/projection_matrix` to be published properly by `/calibration_publisher`. When I run Points Image, I can see the `/points_image` topic in `rostopic list`, but `rostopic echo /points_image` results in this error:

`ERROR: Cannot load message class for [autoware_msgs/PointsImage]. Are your messages built?`

#### Expected Behavior
After running Calibration Publisher, `rostopic echo /camera/projection_matrix` should show the projection matrix being published.

YamatoAndo, 2017-12-25:
@ben-swain Thanks for the report.
Please execute the following command and try again.
```
$ cd <your-Autoware-path>/ros/devel
$ source setup.bash
```

ben-swain, 2017-12-25:
Thanks @YamatoAndo, it works now

closed 2017-12-25 by dejanpan