autowarefoundation/autoware #999: ndt_mapping will crash with gpu option

zyyfly2010, 2017-12-22:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
We want to use ndt_mapping to build the map with GPU.

## Actual Behavior
But it will crash after several minutes. The log is below:
Error: an illegal memory access was encountered /home/xxx/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/NormalDistributionsTransform.cu 929
Error: all CUDA-capable devices are busy or unavailable /home/xxx/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/MatrixDevice.cu 21

But the GPU memory is enough before it crash !
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080  Off  | 00000000:02:00.0  On |                  N/A |
| 17%   50C    P8    22W / 280W |    623MiB / 11169MiB |     23%      Default |
+-------------------------------+----------------------+----------------------+

## Steps to Reproduce the Problem

  1.play a rosbag and publish TF
  2.start ndt_mapping with gpu option
  

## Specifications

  - Ubuntu version:16.04
  - ROS Version: Kinetic
  - Autoware branch: master   Commits on Dec 18, 2017
  - CUDA 8.0

Thanks!

zyyfly2010, 2017-12-22:
And with some other rosbag，it can work well for a while and then crash by another problem as below:

-----------------------------------------------------------------
Sequence number: 3885
Number of scan points: 17609 points.
Number of filtered scan points: 1411 points.
transformed_scan_ptr: 17609 points.
map: 5796352 points.
NDT has converged: 1
Fitness score: 12.4415
Number of iteration: 0
(x,y,z,roll,pitch,yaw):
(279.649, 57.487, 12.5603, -0.201376, 0.193727, 0.252949)
Transformation Matrix:
0.950067 -0.282485  0.132568   279.852
0.245579  0.938977  0.240859   57.8555
-0.192518 -0.196276  0.961464   14.0314
0         0         0         1
shift: 1.20614
-----------------------------------------------------------------
(Processed/Input): (3196 / 3984)
Error: out of memory /home/xxx/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/SymmetricEigenSolver.cu 10
Error: invalid device pointer /home/xxx/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/MatrixDevice.cu 21

amc-nu, 2018-01-27:
this was solved on #1006

closed 2018-01-27 by amc-nu

DevMohamedMI, 2018-07-10:
This issue does not seem to be solved in #1006 

any updates on the illegal memory access GPU problem?

anhnv3991, 2018-07-10:
@DevMohamedMI I thought It was fixed in the latest version of Autoware (both master and develop).
What is your error message?

snuffysasa, 2018-07-11:
@anhnv3991 

I am commit #1189 

And my error is the same as the original poster
Error: an illegal memory access was encountered /home/xxx/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/NormalDistributionsTransform.cu 929
Error: all CUDA-capable devices are busy or unavailable /home/xxx/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/MatrixDevice.cu 21

anhnv3991, 2018-07-15:
@snuffysasa you mean after you commit #1189 you still have that error?

snuffysasa, 2018-07-17:
@anhnv3991 

Correct, I am on that commit, and the ndt mapping gpu problem is still present.

anhnv3991, 2018-07-23:
@snuffysasa well as I see you are using the old code. Just get the latest one and try it again.