autowarefoundation/autoware #1001: error when click map in Runtime Manager

ahuer2435, 2017-12-25:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
it will appear “ok” in “Runtime Manager--->map and next to ref” when click "map"

## Actual Behavior
1. no "ok" appear
2. error info in terminal as follow:
[ERROR] [1514170973.791330698]: Usage:
[ERROR] [1514170973.791367849]: rosrun map_file vector_map_loader [CSV]...
[ERROR] [1514170973.791378036]: rosrun map_file vector_map_loader download [X] [Y]
[ERROR] [1514170973.798526890]: Usage:
[ERROR] [1514170973.798561184]: rosrun map_file points_map_loader noupdate [PCD]...
[ERROR] [1514170973.798577584]: rosrun map_file points_map_loader {1x1|3x3|5x5|7x7|9x9} AREALIST [PCD]...
[ERROR] [1514170973.798587880]: rosrun map_file points_map_loader {1x1|3x3|5x5|7x7|9x9} download
[points_map_loader-3] process has died [pid 25761, exit code 1, cmd /home/yanqiao/workspace/Autoware/ros/devel/lib/map_file/points_map_loader noupdate __name:=points_map_loader __log:=/home/yanqiao/.ros/log/b95cb9f2-e91f-11e7-812f-002324a06c6b/points_map_loader-3.log].
log file: /home/yanqiao/.ros/log/b95cb9f2-e91f-11e7-812f-002324a06c6b/points_map_loader-3*.log
[vector_map_loader-4] process has died [pid 25771, exit code 1, cmd /home/yanqiao/workspace/Autoware/ros/devel/lib/map_file/vector_map_loader __name:=vector_map_loader __log:=/home/yanqiao/.ros/log/b95cb9f2-e91f-11e7-812f-002324a06c6b/vector_map_loader-4.log].
log file: /home/yanqiao/.ros/log/b95cb9f2-e91f-11e7-812f-002324a06c6b/vector_map_loader-4*.log

## Steps to Reproduce the Problem

  1. do as following "https://github.com/ahuer2435/Autoware-Manuals/blob/master/en/Autoware_QuickStart_v1.1.pdf"
  2. step to "3.3 Loading PointCloud and Vector maps"

## Specifications

  - Ubuntu version: ubuntu14.04
  - ROS Version: indigo
  - Autoware branch:master

Thanks.

YamatoAndo, 2017-12-25:
@ahuer2435 Thanks for the report.
Are there .pcd files in ~ /.autoware /data/map/pointcloud_map?
And are there .csv files in ~ /.autoware /data/map/vector_map?

If not found, please download files in step 1.2.
And then, execute the following command of step 1.4 again.
`
~/.autoware $ sh my_launch.sh
`

ahuer2435, 2017-12-25:
@YamatoAndo  
.pcd and .csv files are all exist. step 1.2 already done. as following:
[image]

I find  main() function of points_map_loader.cpp need 3 parameters. and my_map.launch as below:
<launch>

  
  <include file="/home/yanqiao/.autoware/data/tf/tf.launch"/>

  
  <node pkg="map_file" type="points_map_loader" name="points_map_loader" args="noupdate "/>

  
  <node pkg="map_file" type="vector_map_loader" name="vector_map_loader" args=""/>

</launch>

Does the parameters not match?

ahuer2435, 2017-12-25:
[image]

YamatoAndo, 2017-12-25:
hmm... There is not file path in it.
In my environment(Ubutnu16.04 kinetic), as below

```
<launch>

  <!-- TF -->
  <include file="/home/yamato_ando/.autoware/data/tf/tf.launch"/>

  <!-- Point Cloud -->
  <node pkg="map_file" type="points_map_loader" name="points_map_loader" args="noupdate /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00156_-00856.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00156_-00855.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00147_-00850.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00151_-00849.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00159_-00859.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00155_-00854.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00152_-00850.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00150_-00848.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00153_-00850.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00158_-00856.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00148_-00848.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00169_-00868.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00160_-00858.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00153_-00851.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00164_-00863.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00149_-00847.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00151_-00850.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00163_-00862.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00164_-00862.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00153_-00852.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00165_-00864.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00155_-00853.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00168_-00865.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00158_-00858.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00147_-00849.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00151_-00848.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00165_-00863.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00154_-00853.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00866.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00152_-00851.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00148_-00847.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00150_-00846.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00157_-00857.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00161_-00860.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00162_-00862.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00166_-00865.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00161_-00861.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00168_-00867.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00147_-00847.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00156_-00854.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00160_-00860.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00158_-00857.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00865.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00159_-00857.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00864.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00162_-00861.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00159_-00858.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00168_-00868.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00168_-00866.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00155_-00855.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00163_-00861.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00150_-00847.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00160_-00861.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00167_-00867.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00148_-00849.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00155_-00852.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00154_-00852.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00166_-00864.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00152_-00849.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00154_-00851.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00157_-00856.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00147_-00851.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00160_-00859.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00149_-00848.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00149_-00846.pcd /home/yamato_ando/.autoware/data/map/pointcloud_map/bin_Laser-00147_-008 [... cut ...] rk.csv /home/yamato_ando/.autoware/data/map/vector_map/utilitypole.csv /home/yamato_ando/.autoware/data/map/vector_map/curb.csv /home/yamato_ando/.autoware/data/map/vector_map/poledata.csv /home/yamato_ando/.autoware/data/map/vector_map/idx.csv /home/yamato_ando/.autoware/data/map/vector_map/vector.csv /home/yamato_ando/.autoware/data/map/vector_map/stopline.csv /home/yamato_ando/.autoware/data/map/vector_map/roadedge.csv /home/yamato_ando/.autoware/data/map/vector_map/pole.csv"/>

</launch>
```

YamatoAndo, 2017-12-25:
Probably I think that my_launch.sh is not working well.

ahuer2435, 2017-12-25:
Yes, the problem is my_launch.sh. I run my_launch.sh before tar commands. I think this is the problem. I delete the my_launch dir, then rerun my_launch.sh. it work well. thanks @YamatoAndo

closed 2017-12-25 by ahuer2435

Blackshaddock, 2018-08-24:
the .autoware dir is you mkdir, in my work, it has already exist.