autowarefoundation/autoware #1004: [Feature]If the gear is reverse, set speed to negative

YamatoAndo, 2017-12-25:
# New Feature
In the current version, even if it is reverse, speed will be a positive value.
So if the gear is reverse, set speed to negative.

Especially when using can information with ndt_matching, position estimation fails at back.

asimay, 2017-12-27:
I think this may consider the vehicle's dynamic model for reverse back. it's not same with forward dynamic model.

YamatoAndo, 2017-12-27:
I thought that it was not necessary to consider it.
Could you tell me why you think that way?

The code to convert can-information to odometry is included in the following file,
It seems like I have no problems handling forward and backward in the same way.

https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_info_translator/can_info_translator_core.h

amc-nu, 2018-04-07:
@YamatoAndo was this solved with #1004 ? If so, please close this issue. Thanks.

YamatoAndo, 2019-01-23:
Solved #1740

closed 2019-01-23 by YamatoAndo