autowarefoundation/autoware #1009: Build Error with Cuda9

andoh104, 2017-12-31:
# Bug
This issue is same as #910. It was Cuda9 problem.

## Expected Behavior
No Error or Need to limit using ｔｈｅ Cuda9

## Actual Behavior

```
[ 13%] Building NVCC (Device) object computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeFiles/fast_pcl_ndt_gpu.dir/src/fast_pcl_ndt_gpu_generated_Registration.cu.o
In file included from /usr/local/cuda/include/common_functions.h:50:0,
                 from /usr/local/cuda/include/cuda_runtime.h:115,
                 from <command-line>:0:
/usr/local/cuda/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
ptxas info    : 0 bytes gmem
ptxas info    : Compiling entry function 'copyMatrixDevToDev' for 'sm_61'
ptxas info    : Function properties for copyMatrixDevToDev
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 7 registers, 368 bytes cmem[0]
[ 13%] Building NVCC (Device) object computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeFiles/fast_pcl_ndt_gpu.dir/src/fast_pcl_ndt_gpu_generated_VoxelGrid.cu.o
CMake Error at fast_pcl_ndt_gpu_generated_Registration.cu.o.cmake:207 (message):
  Error generating
  /home/tandoh/Autoware/ros/build/computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeFiles/fast_pcl_ndt_gpu.dir/src/./fast_pcl_ndt_gpu_generated_Registration.cu.o

computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeFiles/fast_pcl_ndt_gpu.dir/build.make:84: ターゲット 'computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeFiles/fast_pcl_ndt_gpu.dir/src/fast_pcl_ndt_gpu_generated_Registration.cu.o' のレシピで失敗しました
make[2]: *** [computing/perception/localization/lib/fast_pcl/ndt_gpu/CMakeFiles/fast_pcl_ndt_gpu.dir/src/fast_pcl_ndt_gpu_generated_Registration.cu.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
```

## Steps to Reproduce the Problem

`~/Autoware/ros$ ./catkin_make_release ` l

After removing this macro from CUDA9 header file(/usr/local/cuda/include/crt/common_functions.h), then this error don't occur. 
` #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
`

And this error is very similar to the following:[https://github.com/STEllAR-GROUP/hpx/issues/2815](https://github.com/STEllAR-GROUP/hpx/issues/2815)
I have no idea to fix this.

## Specifications

  - Ubuntu version:16.04
  - ROS Version:Kinetic
  - Autoware branch:master
 - Cuda:9.1

anhnv3991, 2017-12-31:
Install Cuda 8 instead of Cuda 9 can fix that problem.
Maybe some files included in ndt_gpu use the __CUDA_VER__ macro in the ‘if’ command to check the current CUDA version. However, in CUDA 9, that macro is defined as a string, not a number, so compare it with a number in the ‘if’ command is invalid.

andoh104, 2018-01-02:
I can't found __CUDACC_VER__ macro under the Autoware/ros/src, but I found it it in /usr/include/eigen3/Eigen/src/Core/util/Macros.h .
I think Cuda9 and eigen seems to conflict and it is not Autoware's problem.
I found this information :https://devtalk.nvidia.com/default/topic/1023476/boost-under-cuda-9/

I made a temporally patch code to fix this.

kitsukawa, 2018-02-15:
@andoh104 I merged the PR #1027. NDT is working on CUDA 9.0. Thank you for the report.

closed 2018-02-15 by kitsukawa