autowarefoundation/autoware #1012: in new version, when launch runtime manager, always pop out some warning info.

asimay, 2018-01-05:
even when execute > sudo ./run , it will pop out warning info like below:
/home/leon/autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py:2862: wxPyDeprecationWarning: Call to deprecated item. 
  wx.InitAllImageHandlers()

(runtime_manager_dialog.py:4023): IBUS-WARNING **: The owner of /home/leon/.config/ibus/bus is not root!
loading param.yaml
loading qs.yaml
loading setup.yaml

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

amc-nu, 2018-01-28:
You can find the solution to your problem here

https://bugs.launchpad.net/ubuntu/+source/ibus/+bug/1125944

yk-fujii, 2018-01-29:
@amc-nu 
This solution is for ibus, right?
warning of "wxPyDeprecationWarning: Call to deprecated item." is remaining;(

closed 2020-05-29 by gbiggs