autowarefoundation/autoware #1013: Build error

wsung1, 2018-01-06:
Hi, forks,
While doing ./catkin_make_release. I ran into the followings
[ 9%] Built target orb_slam2
make: *** [all] Error 2
Invoking "make -j8" failed.
Almost all related suggestions available here did not work for me.
FYI, I'm using ROS Indigo over Ubuntu 14.04. 
Any idea?
...
ptxas info    : Function properties for normalizeAndTruncate
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 21 registers, 356 bytes cmem[0]
ptxas info    : Compiling entry function 'BilinearKernelTex32F' for 'sm_61'
ptxas info    : Function properties for BilinearKernelTex32F
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 22 registers, 360 bytes cmem[0], 1 textures
[  2%] [  2%] [  2%] Built target cuda_binary_
Built target _vector_map_msgs_generate_messages_check_deps_Wall
Built target _vector_map_msgs_generate_messages_check_deps_RailCrossingArray
Linking CXX shared library /home/futuremobilitylab01/Autoware/ros/devel/lib/libop_utility.so
[  2%] [  2%] Built target _vector_map_msgs_generate_messages_check_deps_CurveMirrorArray
Built target _vector_map_msgs_generate_messages_check_deps_Node
[  2%] [  2%] Built target _vector_map_msgs_generate_messages_check_deps_SideWalkArray
[  2%] Built target _vector_map_msgs_generate_messages_check_deps_WhiteLine
Built target op_utility
[  2%] Built target _vector_map_msgs_generate_messages_check_deps_CrossRoad
[  2%] [  2%] Built target _vector_map_msgs_generate_messages_check_deps_RoadMarkArray
[  2%] [  2%] Built target _vector_map_msgs_generate_messages_check_deps_GutterArray
Built target _vector_map_msgs_generate_messages_check_deps_WallArray
Built target _vector_map_msgs_generate_messages_check_deps_WayAreaArray
[  2%] [  2%] [  2%] Built target _vector_map_msgs_generate_messages_check_deps_Fence
Built target _vector_map_msgs_generate_messages_check_deps_SideStrip
Built target _vector_map_msgs_generate_messages_check_deps_StopLine
[  2%] Built target _vector_map_msgs_generate_messages_check_deps_DriveOnPortionArray
[  2%] Built target _vector_map_msgs_generate_messages_check_deps_RoadSign
[  2%] Built target nav_msgs_generate_messages_lisp
[  2%] Built target _vector_map_msgs_generate_messages_check_deps_ZebraZoneArray
[  2%] Built target tf_generate_messages_cpp
[  2%] Built target _vector_map_msgs_generate_messages_check_deps_DTLaneArray
[  2%] [  2%] Built target _velodyne_msgs_generate_messages_check_deps_VelodynePacket
Built target _velodyne_msgs_generate_messages_check_deps_VelodyneScan
[  2%] Built target tf_generate_messages_lisp
[  2%] Built target actionlib_generate_messages_cpp
[  2%] Built target tf_generate_messages_py
[  2%] Built target actionlib_generate_messages_py
[  2%] [  2%] [  2%] Built target tf2_msgs_generate_messages_py
Built target tf2_msgs_generate_messages_cpp
Built target actionlib_generate_messages_lisp
[  2%] [  2%] Built target tf2_msgs_generate_messages_lisp
Built target nav_msgs_generate_messages_cpp
[  2%] Built target nav_msgs_generate_messages_py
[  2%] Built target _custom_msgs_generate_messages_check_deps_BaroSensorSample
[  2%] [  2%] [  2%] Built target _custom_msgs_generate_messages_check_deps_sensorSample
Built target _custom_msgs_generate_messages_check_deps_positionEstimate
Built target _custom_msgs_generate_messages_check_deps_ImuSensorSample
[  2%] Built target _custom_msgs_generate_messages_check_deps_gnssSample
[  2%] [  2%] Built target _custom_msgs_generate_messages_check_deps_baroSample
Built target _custom_msgs_generate_messages_check_deps_XsensQuaternion
[  2%] Built target _custom_msgs_generate_messages_check_deps_orientationEstimate
[  2%] Built target _custom_msgs_generate_messages_check_deps_GnssSensorSample
[  2%] [  2%] Building CXX object util/packages/RobotSDK/fastvirtualscan/CMakeFiles/fastvirtualscan.dir/FastVirtualScan/fastvirtualscan.cpp.o
Building CXX object util/packages/fake_drivers/CMakeFiles/fake_camera.dir/nodes/fake_camera/fake_camera.cpp.o
[  2%] [  2%] Built target _custom_msgs_generate_messages_check_deps_Internal
Building CXX object sensing/drivers/gnss/packages/garmin/CMakeFiles/garmin_gps_18x_lvc.dir/src/garmin_gps_18x_lvc.cc.o
[  2%] Built target _custom_msgs_generate_messages_check_deps_velocityEstimate
[  2%] Building CXX object sensing/drivers/lidar/packages/hokuyo/CMakeFiles/hokuyo_3d.dir/nodes/hokuyo_3d/hokuyo_3d.cpp.o
[  2%] Built target _kvaser_generate_messages_check_deps_CANPacket
[  2%] Built target visualization_msgs_generate_messages_cpp
[  2%] Built target visualization_msgs_generate_messages_lisp
[  2%] Built target visualization_msgs_generate_messages_py
[... 222 lines cut ...]
/home/futuremobilitylab01/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc:423:50: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::vector<ORB_SLAM2::KeyFrame*>::size_type {aka long unsigned int}’ [-Wformat=]
  fprintf (stderr, "KF: %d\n", allKeyFrames.size());
                                                  ^
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/parameter_container.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_with_hessian.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_levenberg.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/KeyFrameDatabase.cc.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/jacobian_workspace.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel_factory.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel_impl.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/timeutil.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Sim3Solver.cc.o
[  9%] Building C object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/os_specific.c.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/string_tools.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/property.cpp.o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Initializer.cc.o
Linking CXX static library /home/futuremobilitylab01/Autoware/ros/devel/lib/libg2o.a
[  9%] Built target g2o
[  9%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Viewer.cc.o
Linking CXX static library /home/futuremobilitylab01/Autoware/ros/devel/lib/liborb_slam2.a
[  9%] Built target orb_slam2
make: *** [all] Error 2
Invoking "make -j8" failed

wsung1, 2018-01-07:
As advised before, I did 'catkin_make --pkg orb_localizer' (instead of './catkin_make_release') and the result is as follows. Does it indicate that Autoware is well installed?

Base path: /home/futuremobilitylab02/Autoware/ros
Source space: /home/futuremobilitylab02/Autoware/ros/src
Build space: /home/futuremobilitylab02/Autoware/ros/build
Devel space: /home/futuremobilitylab02/Autoware/ros/devel
Install space: /home/futuremobilitylab02/Autoware/ros/install
[  0%] Built target DBoW2
[ 28%] [ 28%] [ 28%] Built target pangolin
Built target std_msgs_generate_messages_py
Built target std_msgs_generate_messages_lisp
[ 28%] Built target imgprocx
[ 28%] Built target std_msgs_generate_messages_cpp
[ 28%] Built target orb_vocabulary
[ 28%] Built target _orb_localizer_generate_messages_check_deps_debug
[ 71%] [ 71%] Built target g2o
[ 71%] Built target orb_localizer_generate_messages_py
Built target orb_localizer_generate_messages_lisp
[ 71%] [ 71%] Built target orb_localizer_generate_messages_eus
Built target orb_localizer_generate_messages_cpp
[ 71%] Built target orb_localizer_generate_messages
[ 85%] Built target orb_slam2
[ 85%] Built target dumpmap
[ 85%] Built target map_publisher
[ 85%] Built target orb_mapping_offline
[ 85%] Built target orb_mapping
[ 85%] Built target orb_matching
[100%] Built target orb_matching_pf
####
#### Running command: "make cmake_check_build_system" in "/home/futuremobilitylab02/Autoware/ros/build"
####
####
#### Running command: "make -j8 -l8" in "/home/futuremobilitylab02/Autoware/ros/build/computing/perception/localization/packages/orb_localizer"
####

sujiwo, 2018-01-07:
This means that the error does not lie in orb_localizer.

wsung1, 2018-01-07:
@sujiwo Thanks to your kindness, I have just resolved this problem. Its cause is unclear so I reinstalled the newest version of Autoware from scratch. Thanks, anyway.

closed 2018-01-07 by wsung1