autowarefoundation/autoware #1016: Points2image process dying, related to /camera

ben-swain, 2018-01-09:
# Bug
## Expected Behavior
I expect the points2image process to not die when I run a node which subscribes to topic /camera/image_raw or when I use `rostopic echo /camera/image_raw`.

## Actual Behavior
After running `roslaunch runtime_manager points2image.launch`, topic /points_image is published correctly. But when I run my /camera/image_raw subscriber node or `rostopic echo /camera/image_raw`, the points2image process dies with the following error:

> [points2image-1] process has died [pid 89528, exit code -11, cmd /home/shared/Autoware/ros/devel/lib/points2image/points2image __name:=points2image __log:=/home/team0/.ros/log/5b772bf8-f4c1-11e7-b8e5-ac1f6b056614/points2image-1.log].
> log file: /home/team0/.ros/log/5b772bf8-f4c1-11e7-b8e5-ac1f6b056614/points2image-1*.log" when running roslaunch runtime_manager points2image.launch

I could not find the log file in that location.

## Steps to Reproduce the Problem

  1. Download [THIS](https://drive.google.com/file/d/1Dfz0sLWXDYm7FI_9xOd4I9xXrp6Y3b5A/view?usp=sharing) rosbag which contains topics: /camera/image_raw, /camera_info, /points_raw, and /projection_matrix.
  2. Play the rosbag looped: `rosbag play --loop (path)`
  3. Run `roslaunch runtime_manager points2image.launch`
  4. The points2image process fails ("[points2image-1] process has died")

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: Kinetic
  - Autoware branch: master

ben-swain, 2018-01-09:
The problem was solved by changing the way my pointgrey cameras launch. I can add more details tomorrow. I'm not sure why it fixed the problem.

closed 2018-01-28 by amc-nu

rlfgus777, 2018-10-03:
I solved that problem, removing the carmeraInfo data in Raw_bag file. 
camera info topic is published by calibration publisher. So camera_info topic in Raw_bag file is overlapped with calibration publisher's camera_info.