autowarefoundation/autoware #1017: Vectormap is not displayed in RViz 1.12.14

yk-fujii, 2018-01-09:
# Bug
The Vectormap message not accepted by RVIZ due to the following PR.
https://github.com/ros-visualization/rviz/pull/1167
Normalization of quaternions is required.

[image]

## Specifications
  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - RVIZ Version: 1.12.14(released at 2017-12-19)
  - Autoware branch: master

yk-fujii, 2018-01-09:
Other nodes that use marker seem to be similar.

aohsato, 2018-01-09:
This occurred by not normalized quaternion in vizualization_msgs::Marker.
When we call Marker constructor, then quaternion filled by zero is created.
For now, I added this line on each node publishing Marker, so then it's solved.
```
marker.pose.orientation.w = 1.0
```

yk-fujii, 2018-01-10:
https://discourse.ros.org/t/fix-released-rviz-1-12-14-not-displaying-markers-etc-with-invalid-quaternions/3610

closed 2018-01-10 by yk-fujii