autowarefoundation/autoware #1020: SSD node cannot be launched

MadCauchy, 2018-01-09:
Hello Autoware,

i followed the instruction of ssdcaffe installation. And i choose to use the trained model VOC0712. I downloaded the package into the model folder of ssd. Then I try to use the Runtimemanager to launch the ssd node, but it failed. The error message is as following:

> F0109 14:28:00.256325 27615 io.cpp:45] Check failed: fd != -1 (-1 vs. -1) File not found: data/VOC0712/labelmap_voc.prototxt
> *** Check failure stack trace: ***
> [ssd-1] process has died [pid 27615, exit code -6, cmd /home/adp/Autoware/ros/devel/lib/cv_tracker/ssd_unc __name:=ssd __log:=/home/adp/.ros/log/243ef1aa-f540-11e7-8b47-e094679bb02c/ssd-1.log].
> log file: /home/adp/.ros/log/243ef1aa-f540-11e7-8b47-e094679bb02c/ssd-1*.log
> process[ssd-1]: started with pid [27615]
> all processes on machine have died, roslaunch will exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

I have fund a similar issue with coco model.  #958

## Specifications

  - Ubuntu version:16.04
  - ROS Version:kinetic
  - Autoware branch: master

Thank you!

amc-nu, 2018-01-27:
You're using a different pre trained model version than the required by the node. 

Please use the pre trained models from the version indicated in the README file of the ssd node.

closed 2018-01-27 by amc-nu

sjtugjy, 2018-06-07:
@amc-nu 

Hi,
It seems there is no VGG_VOC0712_SSD_500x500_iter_60000.caffemodel anymore on https://github.com/weiliu89/caffe/tree/ssd#models.

If using this model (https://gist.github.com/weiliu89/2ed6e13bfd5b57cf81d6) which is 500x500,
there is no output in the RVIZ.

Koto-Norose, 2018-06-17:
>F0109 14:28:00.256325 27615 io.cpp:45] Check failed: fd != -1 (-1 vs. -1) File not found:data/VOC0712/labelmap_voc.prototxt

Change the label_map_file field of the deploy.prototxt solved the problem.  

Change from relative path
>data/VOC0712/labelmap_voc.prototxt

to the absolute path.
>/home/hoge/ssdcaffe/data/VOC0712/labelmap_voc.prototxt

Koto-Norose, 2018-06-17:
>F0109 14:28:00.256325 27615 io.cpp:45] Check failed: fd != -1 (-1 vs. -1) File not >found:data/VOC0712/labelmap_voc.prototxt  

Change the label_map_file field of the deploy.prototxt solved the problem. 

Change from relative path  
>data/VOC0712/labelmap_voc.prototxt  

to the absolute path.  
>/home/hoge/ssdcaffe/data/VOC0712/labelmap_voc.prototxt
--

kttnk, 2018-06-17:
>If using this model (https://gist.github.com/weiliu89/2ed6e13bfd5b57cf81d6) which is 500x500,
there is no output in the RVIZ.

I used the 300x300 caffe model from ( https://github.com/weiliu89/caffe/tree/ssd#models ) for car detection, and seems it's working.

models/VGGNet/VOC0712Plus/SSD_300x300_ftVGG_VOC0712Plus_SSD_300x300_ft_iter_160000.caffemodel