autowarefoundation/autoware #1021: How to add new drivers for camera and lidar in GUI

MadCauchy, 2018-01-09:
Hi, Autoware

I'm trying to use the camera and Lidar in my lab with Autoware system.  I have already found the correct drivers for camera GMSL and Lidar iBeo Lux4. I want to add these two drivers into GUI runtimemanager. But I don't know how. I have tried add some lines in Autoware/ros/src/util/packages/runtime_manager/scripts/sensing.yaml. Although two icons are correctly added, the node cannot be launched.  Can you please give me a suggestion, how can I add these two options into RTM. 

Thanks a lot!

  - Ubuntu version:16.04
  - ROS Version:Kinetic
  - Autoware branch:master

yk-fujii, 2018-01-15:
Could you run your node by using rosrun/roslaunch commands (or a command you added to the "run" line)?
The runtime manager executes the command described in "run" line.

e.g.
https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/runtime_manager/scripts/sensing.yaml#L27

wsung1, 2018-01-15:
@MadCauchy In fact, I have a similar problem with all different sensors. Would you share your result from following instructions from @yk-fujii? It will be greatly appreciated.

MadCauchy, 2018-01-18:
I can use this command to start the node of camera: 

`roslaunch gmsl_n_cameras gmsl_n_cameras.launch`

I insert the following lines to the file sensing.yaml

```
       - name : gmsl
         desc : gmsl desc sample
         probe:
         run  : roslaunch gmsl_n_cameras gmsl_n_cameras.launch
```

The option shows in the Runtime_Manager. But it did't work. The error message looks like this:

> [gmsl_n_cameras.launch] is neither a launch file in package [gmsl_n_cameras] nor is [gmsl_n_cameras] a launch file name
> The traceback for the exception was written to the log file

Then I tried to add a source command

```
  - name : gmsl
         desc : gmsl desc sample
         probe:
         run  : source ~/workspace/catkin_ws/devel/setup.bash
         run  : roslaunch gmsl_n_cameras gmsl_n_cameras.launch
```

It didn't work either. the error message is just exactly like before.

I also want to ask, what the probe line means? How can I add a source command correctly in it? 

Thank you a lot!

amc-nu, 2018-01-28:
seems your driver is in a different workspace. The easiest solution I can think is to move your package to the same Autoware workspace, re compile and roslaunch.

closed 2018-01-28 by amc-nu