autowarefoundation/autoware #1024: Update drive-px2 dependencies

aohsato, 2018-01-11:
# Bug
For now, the versions of SSL package was updated from 4.9 to 4.10.
https://github.com/CPFL/Autoware/blob/develop/docs/en/installation_with_drivepx2.md

```
sudo apt-get install -y libssl1.0.0=1.0.2g-1ubuntu4.9
sudo apt-get install -y libssl-dev=1.0.2g-1ubuntu4.9
```

Finally, how do you think to deal with continuous updates of dependencies like this?

  - Ubuntu version: 4.9.38-rt25-tegra
  - ROS Version: kinetic
  - Autoware branch: develop

dejanpan, 2018-01-15:
@aohsato you need to have an Autoware branch and CI pipeline that builds for ARM64 and their version of Ubuntu. CI has to always start building from a clean and latest Ubuntu version.

I suggest to have 1h workshop with Apex and we can show you how CI is done correctly.

yk-fujii, 2018-01-20:
First of all we will add docker. #1037

closed 2020-05-29 by gbiggs