autowarefoundation/autoware #1031: how is the transformation from map to base_link calculated ?

dlounis, 2018-01-16:
where is the transformation between map frame and base_link calculated ? Which process provides this transformation ?

When I perform localization based on AW tutorial (tierV) the lidar scans do not overlap the map generated earlier by ndt mapping.

Thanks.

kitsukawa, 2018-01-17:
@dlounis ndt_matching publishes transformation between map frame and base_link.
Can you follow the instruction in the video below?
https://www.youtube.com/watch?v=ODlxMzGTJzw

dlounis, 2018-01-17:
@kitsukawa Thanks. It is not easy to initialize the position manually therefore I used the GNSS in ndt_matching (together with nmea2tfpose node).
I managed to make it run correctly with this setup using PCD files provided in the moriyama sample. However, when I use the PCD files from the same sample that I have generated using approximate_ndt_mapping, and following the same procedure mentioned before for localization using GNSS, I cannot display the map in the world frame. Do you have an idea on why I am having this behavior ?

Thanks.

dlounis, 2018-01-17:
After investigating deeper it looks like the generated map and the lidar scans when displayed on RVIZ during localization, are not projected on the same frame. A frame transformation is missing from the lidar scans (points_raw) and the map points. I don't know how to deal with that so far.

dlounis, 2018-01-19:
any help ?

zhuxuekui3, 2018-03-08:
dlounis , i have same problem, do you have solved?

zhuxuekui3, 2018-03-09:
@dlounis  can you contact me (my weixin number is 18813069635 )

wsung1, 2018-04-19:
@amc-nu,
You told that a tf between the GNSS coordinates and the exact point where you started to create the map is  needed. Would you tell me how to make this tf?