autowarefoundation/autoware #1034: not get a normal path using astar_planner

gongyue666, 2018-01-18:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
I want to get a smoother shortest path with hybrid A* using my robot.But now the path I get not good in a test environment in my computer.

## Actual Behavior
The path have many curves where I think it should be straight line .No obstacles between start and target pose but the path has many small curves.

## Steps to Reproduce the Problem
 here is my computer test environment:
  1.<launch>
 <node name="tf_broadcaster" pkg="hybrid_astar" type="tf_broadcaster" />
 <node name="map_server" pkg="map_server" type="map_server" args="$(find hybrid_astar)/maps/map.yaml" />
</launch>
(The map is empty with no obstacles.)
  2.<launch>
  <arg name="use_2dnav_goal" default="true" />
  <arg name="map_frame" default="map" />
  <arg name="angle_size" default="48" />
  <arg name="minimum_turning_radius" default="0.05" />
  <arg name="obstacle_threshold" default="70" />
  <arg name="robot_length" default="0.50" />
  <arg name="robot_width" default="0.30" />
  <arg name="base2back" default="0.15" />
  <arg name="curve_weight" default="1.00" />
  <arg name="reverse_weight" default="2.50" />
  <arg name="distance_heuristic_weight" default="1.0" />
  <arg name="potential_weight" default="10.0" />

  <arg name="lateral_goal_range" default="0.1" /> 
  <arg name="longitudinal_goal_range" default="0.1" /> 
  <arg name="goal_angle_range" default="1.0" /> 
  
  <arg name="time_limit" default="5000.0" />
  <arg name="use_back" default="false" />
  <arg name="use_wavefront_heuristic" default="false" />
  <arg name="use_potential_heuristic" default="false" />
  <arg name="publish_marker" default="true" />

  
  <arg name="obstacle_detect_count" default="8" />
  <arg name="avoid_distance" default="1.3" />
  <arg name="avoid_velocity_limit_mps" default="0.45" />
  <arg name="upper_bound_ratio" default="1.20" />
  <arg name="avoidance" default="true" /> 
  <arg name="switch_path" default="true" /> 
  
  <node pkg="astar_planner" type="obstacle_avoid" name="obstacle_avoid" output="screen">
    <param name="use_2dnav_goal" value="$(arg use_2dnav_goal)" />
    <param name="map_frame" value="$(arg map_frame)" />
    <param name="angle_size" value="$(arg angle_size)" />
    <param name="minimum_turning_radius" value="$(arg minimum_turning_radius)" />
    <param name="obstacle_threshold" value="$(arg obstacle_threshold)" />
    <param name="use_back" value="$(arg use_back)" />
    <param name="robot_length" value="$(arg robot_length)" />
    <param name="robot_width" value="$(arg robot_width)" />
    <param name="base2back" value="$(arg base2back)" />
    <param name="curve_weight" value="$(arg curve_weight)" />
    <param name="reverse_weight" value="$(arg reverse_weight)" />
    <param name="distance_heuristic_weight" value="$(arg distance_heuristic_weight)" />
    <param name="potential_weight" value="$(arg potential_weight)" />
    <param name="time_limit" value="$(arg time_limit)" />
    <param name="lateral_goal_range" value="$(arg lateral_goal_range)" />
    <param name="longitudinal_goal_range" value="$(arg longitudinal_goal_range)" />
    <param name="goal_angle_range" value="$(arg goal_angle_range)" />
    <param name="use_wavefront_heuristic" value="$(arg use_wavefront_heuristic)" />
    <param name="use_potential_heuristic" value="$(arg use_potential_heuristic)" />
    <param name="publish_marker" value="$(arg publish_marker)" />

    <param name="obstacle_detect_count" value="$(arg obstacle_detect_count)" />
    <param name="avoid_distance" value="$(arg avoid_distance)" />
    <param name="avoid_velocity_limit_mps" value="$(arg avoid_velocity_limit_mps)" />
    <param name="upper_bound_ratio" value="$(arg upper_bound_ratio)" />
    <param name="avoidance" value="$(arg avoidance)" />
    <param name="change_path" value="$(arg switch_path)" />
  </node>

  
</launch>

## Specifications

  - Ubuntu version:16.04
  - ROS Version:Kinect
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

closed 2020-05-29 by gbiggs

rosliwanlei, 2020-12-01:
do you have make the waypoints using the hybrid A*? i donot make it! could you help me?

JWhitleyWork, 2020-12-01:
@rosliwanlei Please use [ROS Answers](https://answers.ros.org/) with the `autoware` tag for support questions.