autowarefoundation/autoware #1035: ndt_mapping then ndt_matching

dlounis, 2018-01-18:
I want to perform localization using a map that was generated with approximate_ndt_mapping (pcd files).
I managed to generate such a map (both for kitti data as well as moriyama sample). I follow the steps provided in autoware documentation (tierV). 

The issue I have is that when I perform ndt_matching, on RVIZ the map and the points_raw do not overlap because they are not on the same frame.

If I chose in the global setting Map as frame I can only see the map. I have modified the tf.launch file so that the transform from map to world is identity. But this doesn't fix my issue.

I have played around with transformations for this end but it does not seem to work.

I am missing something and I don't know what.

Also, I have noticed that for the provided moriyama pcd map, on RVIZ the map can only be displayed in the world frame which makes it possible to have the lidar scans (points_raw) overlap the map. However, when it comes to the pcd files generated with approximate_ndt_mapping, the map can only be seen on the Map frame (not on the world frame) while the scans are on another frame (world).

Your help will be very appreciated.

Thanks.

## Specifications

  - Ubuntu version:14.04
  - ROS Version:Indigo
  - Autoware branch:Master

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

amc-nu, 2018-01-28:
Please follow the localization tutorial (https://github.com/CPFL/Autoware-Manuals/tree/master/en). Very probably you are not providing a good initial position.
NDT connects the two frames thought a transformation in tf

dlounis, 2018-01-29:
Hello,

thanks for your reply. I have followed the tutorials carefully several times. I provide an initial pose that is 0 0 0 0 0 0. With the same setup, when I am using the pcd's provided in the quick start demo, I manage to perform localization with ndt matching without problems. The problem occurs when I use the pcd's that I've generated with approximate_ndt_mapping. I have noticed that the provided pcd's from quick start are less dense than the ones that I generate with approximate_ndt_mapping (although I use filter resolution of 0.2 and do not activate "original" in app tab of approximate_ndt_mapping).

Regards.

Pallav1299, 2020-04-17:
The pcd map in the sample_moriyama demo data is also in the map frame. You can check this by reading the PCD file converted to ASCII format. It's just that the pcd map is transformed and their is a transform between the map & world frames. 
This may be to reduce the number of static transform declarations in the launch file.