autowarefoundation/autoware #1038: ndt_matching : map points not transformed in world frame

dlounis, 2018-01-20:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug ndt_matching : map points not transformed in world frame
## Expected Behavior
map and ponts_raw should overlap (be in the same frame)

## Actual Behavior
points_raw are seen in the world frame
map points are seen in the map frame (far from world frame)

## Steps to Reproduce the Problem

  1. generate pcd submaps using ndt_mapping (moriyama bag file is used to get points_raw : lidar scans)
  1. pcd files are loaded in map tab + tf.launch 
  1. play moriyama bag file + ndt_matching + rviz

## Specifications

  - Ubuntu version:14
  - ROS Version:indigo
  - Autoware branch:master

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

kitsukawa, 2018-01-22:
@dlounis Which tf.launch do you use? The PCDs created by ndt_mapping/approximate_ndt_mapping has local coordinate(not global coordinate) so you need to load tf_local.launch(in Autoware/ros/src/.config/tf/ directory).

dlounis, 2018-01-22:
@kitsukawa Hello, I have tried the tf_local.launch following your instruction and still have the issue of not seeing points_raw on the map. I have already tried the identity transform before and I have noticed that the world and map frames overlap (which is exactly what the tf_local.launch does).

The points_raw evolve on another frame very far from the map/world frames. Below are the outputs from tf_echo : 

rosrun tf tf_echo world map
At time 1427158073.413
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

rosrun tf tf_echo world base_link
Failure at 1427157702.295665655
Exception thrown:Could not find a connection between 'world' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
The current list of frames is:
Frame velodyne exists with parent base_link.
Frame map exists with parent world.

Failure at 1427157702.295665655
Exception thrown:Could not find a connection between 'world' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
The current list of frames is:
Frame velodyne exists with parent base_link.
Frame map exists with parent world.

At time 1427157703.198
- Translation: [9431.729, 32028.758, 6129.252]
- Rotation: in Quaternion [-0.052, -0.182, 0.908, 0.373]
            in RPY (radian) [-0.379, -0.042, 2.371]
            in RPY (degree) [-21.723, -2.383, 135.850]

I don't understand what am I doing wrong ?

Your help is very appreciated.

kitsukawa, 2018-01-22:
@dlounis Can you send me the rosbag and PCDs?

dlounis, 2018-01-22:
@kitsukawa Ok. I will send you the files. The rosbag is the moriyama sample : sample_moriyama_150324 : 

http://db3.ertl.jp/autoware/sample_data/sample_moriyama_150324.tar.gz

the pcd's are in my dropbox : 
https://www.dropbox.com/s/00zu44td665sry4/mori10_50_noimu.zip?dl=0

Please let me know if you experience any problem in downloading from my dropbox.

Thank you.

dlounis, 2018-01-23:
@kitsukawa Hello, have you had a chance to look into my pcd's ?

Thanks.

kitsukawa, 2018-01-31:
@dlounis I checked you data and localization worked correctly. But I downsampled the PCDs because it is very dense and is heavy to visualize on RViz. So please downsample the points of all the PCDs using PCD Filter in Map tab first. 0.2 will be fine for Leaf Size.

dlounis, 2018-01-31:
Hi @kitsukawa. Thanks a lot for your answer, this helps. Since I posted this questions I went through many trials. I have noticed that I have sent you the original rosbag (the one provided by Autoware for its QS example: sample_moriyama_150324) instead of generating a new rosbag containing only points_raw (as explained in the tutorials). Is there a difference in using one or the other rosbag files ?

For instance, I have generated a rosbag from the sample_moriyama_150324 where I record the points_raw only. But when I try to display them in rviz in the base_link frame (I apply the tf that is in the setup tab), I get an error saying : message removed because it is too old. Is the generated rosbag corrupt ?

In order to record a new rosbag from another rosbag do we need to click on the play button of the rosbag player together with the Start button of the rosbag recorder ? If it is the case, there will be a delay between recording and playing the rosbags. Can this be a reason for a corrupt generated rosbag ?

Thanks a lot for your help :)

dlounis, 2018-01-31:
@kitsukawa another question regarding the filtered map. Why can I not get downsampled maps when using approximate_ndt_mapping ? even if I activate downsampling in the app tab of approximate_ndt (I can only get downsampling working with ndt_mapping, but I need to use the approximate_ndt version because I have a long distance map) ?

Thanks again.

dlounis, 2018-01-31:
@kitsukawa I tried to activate PCD filter but it does not work. I am following Autoware pdf tutorial (it says to indicate the location of the pcd files in the ref case). Please, what shall I enter as argument in REF ? Shall I activate the point cloud as well ? It is not very clear.

kitsukawa, 2018-01-31:
@dlounis 
1. You don't need to create new rosbag. You can create maps and can localize in the maps using same rosbag.
2. approximate_ndt_mapping only outputs original (not downsampled) PCDs. So you need to downsample afterwards.
3. Click `Ref` and select PCDs that you want to downsample, then click `PCD Filter`. The downsampled PCDs (named 0.200_***.pcd or something) will be output in the same directory as inout PCDs.

closed 2018-02-20 by kitsukawa

wsung1, 2018-04-18:
@kitsukawa and @dlounis,
So, the fact that @dlounis didn't downsample his pcd files (generated by approximate_ndt_mapping) turned out to be the root cause that @dlounis couldn't have his map and base_link frames overlaid?
Do I understand @dlounis's problem and @kitsukawa's solution correctly?

HongweiSunny, 2019-03-09:
Actually, I have met this problem too. And it confused me a lot for days. I can not see the points_raw in rviz like the original problem above. Need help. QAQ.