autowarefoundation/autoware #1046: Questions about the calibration part.

ttomchy, 2018-01-25:
I am currently learning how to calibrate the camera and the velodyne . Now I am using the the  bag file which is provided. However when I trying to extract the laser points I can only select one line  of the points which is very less points. I don't know how to select more points as shown on the video ( https://www.youtube.com/watch?v=pfBmfgHf6zg) . It shows that they can select more points on the checkboard plane.   I wonder whether it will effect the calibration results. Or is there any tips about how to use the calibration tool.                                                                                                             

[image]

amc-nu, 2018-01-25:
Hello @ttomchy 
Thanks for your interest
Please refer to the wiki article about calibration. You can find your answer there.
https://github.com/CPFL/Autoware/wiki/Calibration

wsung1, 2018-01-26:
Hi, @ttomchy @amc-nu,
Thanks for providing https://github.com/CPFL/Autoware/wiki/Calibration.
As a preparation for calibration, it says
1.1　Activate the camera.
　　　*Set the color and resolution same as Autoware.
1.2　Activate Velodyne.
Would you please tell me how to activate the camera and Velodyne (before 1.3 Run Autoware)?
I've actually observed such an error and Is this because I didn't activate Velodyne?

[image]

closed 2018-01-28 by amc-nu