autowarefoundation/autoware #1049: autoware frames

dlounis, 2018-01-25:
Hello,

please, can someone tell me what does the frame Mobility correspond to ?

Also, I wonder why is the frame World situated at (14771 84757 -39) from Map frame ?

Why are the map and the lidar scans in the moriyama example displayed in the World frame instead of the Map frame ?

Thanks a lot for your help.

amc-nu, 2018-01-27:
@dlounis Please disregard the `mobility` frame. Is not used.

Those coordinates are based on a Japanese method to localize in Japan.

`world` is a "local" frame (an intermediate one used by Autoware) used to avoid using huge numbers from the `map` frame. 
http://www.ros.org/reps/rep-0105.html

closed 2018-01-27 by amc-nu

dlounis, 2018-01-27:
Thanks.
I still don't understand why the map is so far from the map frame (14771 84757 -39) ?

When I try to perform localization whether it is with moriyama data or with Kitti data, NDT matching does not work because the map on rviz is displayed on a separate frame from the lidar scans. I have asked this question before but the answer I got was to apply an identity transform from world to map, but unfortunately this did not solve the problem which is to have the map and the lidar scans on separate frame.

Thanks a lot for your help.

amc-nu, 2018-01-30:
Once you provide an initialization position for NDT using the 2D Pose Estimate tool, NDT will add the transformation to the tf tree, rviz will automatically show the map and sensor points, just select "world" in the fixed frame in rviz. You can run RQT and see the tf tree.

Regarding the numbers, imagine a huge map of the whole world. Where is the origin? We can choose whatever we want. It is arbitrary, right? Having those big numbers, allow to localize all the areas in the world using the same coordinate frame.

In this case, Autoware is using the standard defined by Japan's Geospatial Information Authority.