autowarefoundation/autoware #1050: cmd_param calibration is required

wsung1, 2018-01-26:
Hello all,
By checking the type of LiDAR being connected to PC, the Runtime Manager said "cmd_param calibration is required" as shown below. Does anybody know what this means?
[image]

closed 2018-01-26 by wsung1

amc-nu, 2018-01-30:
@wsung1 you need to select the corresponding calibration file to your device:
https://github.com/CPFL/Autoware/tree/master/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_pointcloud/params

wsung1, 2018-01-30:
Thanks for your reply @amc-nu,
By any chance, do you know the difference between VLP16db and VLP16_hires_db?