autowarefoundation/autoware #1051: cannot launch node of [grasshopper3_camera]

wsung1, 2018-01-26:
Hi, Autoware,
Following the Autoware's setup, I'm using a grasshopper3. It is properly installed and this is checked by FlyCapture2, as shown below.
[image]
However, when I roslaunch autoware_pointgrey_drivers grasshopper3.launch, an error occurs
[image]
If I roslaunch it by checking PointGrey Grasshopper3 in Sensing\Runtime Manager, the same error comes up
[image]
Is there anybody who know what this happens?
Thanks for reading :)

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: Kinetic
  - Autoware branch: 1.6.0

wsung1, 2018-01-27:
Dear @amc-nu,
If you don't mind, please look over #1051. It might be just a piece of cake to you.
Thank you.

amc-nu, 2018-01-27:
can you please show the real error?. it might be located in the ROS log. There are no details in the screenshot you shared, or run with screen output, so the actual error is shown.
http://wiki.ros.org/roslaunch/XML/node

wsung1, 2018-01-27:
Thanks for your reply, @amc-nu!
The result from --screen option is as follows.
[image]
[roslaunch-futuremobilitylab01-HP-EliteDesk-800-G3-TWR-5276.log](https://github.com/CPFL/Autoware/files/1669840/roslaunch-futuremobilitylab01-HP-EliteDesk-800-G3-TWR-5276.log)
I hope these could help you figure out.

amc-nu, 2018-01-27:
the grasshopper node does not exist. It was not compiled by catkin because you need to install the PointGrey SDK.

https://github.com/CPFL/Autoware/blob/master/ros/src/sensing/drivers/camera/packages/pointgrey/CMakeLists.txt#L22

wsung1, 2018-01-27:
Thanks, @amc-nu!
I've just installed PointGrey SDK - flycapture2-2.11.3.425-amd64.
The pointgrey CMakeLists.txt above says it is expected to be installed to /usr/include/flycapture.
I think it is installed in a correct path, which can be backed by
[image]
Then, I've catkin-built again by
~/Autoware/ros$./catkin_make_release
, resulting in
[compile_out.log](https://github.com/CPFL/Autoware/files/1669872/compile_out.log)
It seems that the grasshopper node was compiled in this time.
Finally, I roslaunch again in the same manner, which produces a different error
[image]
[roslaunch-futuremobilitylab01-HP-EliteDesk-800-G3-TWR-29008.log](https://github.com/CPFL/Autoware/files/1669877/roslaunch-futuremobilitylab01-HP-EliteDesk-800-G3-TWR-29008.log)
I'm so sorry for bothering you, @amc-nu, but please take care of this, if you have time to do.
I'd appreciate you :)

amc-nu, 2018-01-27:
Again there’s no error in the log file you sent. It points to a different log file.

wsung1, 2018-01-27:
@amc-nu, is a log file you need to look into grasshopper-2.log?
I don't know why but it doesn't exist in the folder given, as shown below.
[image]

wsung1, 2018-01-27:
@amc-nu, please let me know how to retrieve the log file named grasshopper.
Once again, I'd like to apologize to you for all these troubles I made.

closed 2018-01-29 by wsung1

amc-nu, 2018-01-30:
can you try running rosrun instead of roslaunch? that will give the complete output.

wsung1, 2018-01-30:
Sure, I'd like to do and also I think I should do! But on the one hand, I feel embarrassed because my fault is too rudimentary.
I still don't know how to access such log files (grasshopper-2.log) as they don't exist in the directory given.
However, I've got to know what causes the processes to die.In my case, this was related to configuring USBFS. I should've done this beforehand but I didn't. Refer to https://www.ptgrey.com/tan/10685
I sincerely appreciate @amc-nu 's supports!

GeoBIMpro, 2018-01-30:
@wsung1 Which Lens are you using for Grasshopper3?(Model of C Mount Lens)

wsung1, 2018-01-30:
Hi, @GeoBIMpro! Just out of the curiosity? As attached, I've had 3 different lenses (all C-mount). Perhaps M1614 will be used for Autoware. This never means it is best suited to Grasshopper3 or something.
I hope Autoware would speak out their camera and lens specifications in detail. I think this can be reference to a novice developer like me.