autowarefoundation/autoware #1052: c++: internal compiler error: Killed (program cc1plus)

lakeprince, 2018-01-27:
[  1%] Built target op_utility
Scanning dependencies of target fast_pcl_registration
[  1%] Building CXX object computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/registration.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
computing/perception/localization/lib/fast_pcl/filters/CMakeFiles/fast_pcl_filters.dir/build.make:110: recipe for target 'computing/perception/localization/lib/fast_pcl/filters/CMakeFiles/fast_pcl_filters.dir/src/voxel_grid_covariance.cpp.o' failed
make[2]: *** [computing/perception/localization/lib/fast_pcl/filters/CMakeFiles/fast_pcl_filters.dir/src/voxel_grid_covariance.cpp.o] Error 4
CMakeFiles/Makefile2:377: recipe for target 'computing/perception/localization/lib/fast_pcl/filters/CMakeFiles/fast_pcl_filters.dir/all' failed
make[1]: *** [computing/perception/localization/lib/fast_pcl/filters/CMakeFiles/fast_pcl_filters.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Building CXX object computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/correspondence_estimation.cpp.o
[  2%] Building CXX object computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/correspondence_types.cpp.o
[  2%] Building CXX object computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/icp.cpp.o
[  2%] Building CXX object computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/icp_nl.cpp.o
[  2%] Building CXX object computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/ndt.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/build.make:158: recipe for target 'computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/icp_nl.cpp.o' failed
make[2]: *** [computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/src/icp_nl.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:3595: recipe for target 'computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/all' failed
make[1]: *** [computing/perception/localization/lib/fast_pcl/registration/CMakeFiles/fast_pcl_registration.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed
➜  ros git:(master) 
===================================
Parallel Desktop for Mac Ubuntu 16.04
ROS Kinetic

closed 2018-01-27 by lakeprince

lakeprince, 2018-01-27:
i have solved this problem, it is the ram is not enough, especially in a virtual machine.

baaixw, 2020-01-17:
yes, exactly