autowarefoundation/autoware #1059: Point cloud visible with RViz but invisible with calibration toolkit

wsung1, 2018-01-30:
Dear all,
For sensor calibrating, I have point cloud data "visible" through RViz
screenshot from 2018-01-30 13-11-00
but "not visible" on calibration toolkit.
screenshot from 2018-01-30 13-10-44
As shown above, I changed the background color but still I couldn't see any points. Thus, I think I have to work on GLViewer but I can't figure out how to do exactly.
This problem is related to #498. Here, talks between @Eric-Gonzalez and @alexanderhmwand could give a clue but unfortunately I'm new to GLViewer.
e.g.) (2) The GLViewer derived from the QGLWidget doesn't work? <1% possiblility~ (line 714 - 730)
I don't know what happens to the Qt's GLWidget. But if the Velodyne data is correctly loaded, the only problem is the visualization conducted by OpenGL. Please check the OpenGL version >=2.0, and test the GLViewer library (https://github.com/CPFL/Autoware/tree/master/ros/src/util/packages/RobotSDK/glviewer) in a simple test application, and also test the "glDrawArrays" function.
e.g.) The default GLViewer perspective was oriented away from the points. Using a combination of the WASD, and QE keys I was able to bring the points into view.
If you don't mind, would you please elaborate it for me?
Thank you so much!

wsung1, 2018-01-31:
@amc-nu, I'm sorry to keep bothering you but by any chance do you have any opinion on this?

alexanderhmw, 2018-02-02:
Did you try to use WASD and QE to change the pose of the viewport?
If you click the 3D viewport and push 'S' key to move backward, you will see an RGB XYZ-axis. That means the GLViewer works fine. Then please change the background to check where the points are.
Thanks!

amc-nu, 2018-03-06:
Did you fix this @wsung1 ?
If not can check if your point cloud is being published in /points_raw?

wsung1, 2018-03-06:
Done thanks to @alexanderhmw.
Thanks, @amc-nu!

closed 2018-03-06 by wsung1

QipiaoZhou, 2022-06-26:
> Did you try to use WASD and QE to change the pose of the viewport?
> If you click the 3D viewport and push 'S' key to move backward, you will see an RGB XYZ-axis. That means the GLViewer works fine. Then please change the background to check where the points are.
> Thanks!

after pushing “s”，i canot see xyz axis，what can i do then