autowarefoundation/autoware #1072: [autoware] openni2 not found VTK conflict

dlounis, 2018-02-06:
After building Autoware in Ubuntu 16.04 and Ros Kinetic, I managed to run the quick start example. However I have noticed that what I get on my screen misses some details when compared to what is shown on the Quick start pdf.

I have checked the build log and I noticed those two messages :

-No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS)
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

I have installed OpenNI2 with these instructions https://codeyarns.com/2015/08/04/how-to-install-and-use-openni2/ but the message on missing openni2 is still there. Also I have looked into a solution for the message regarding VTK but I did not manage to fix the problem (I have installed python-vtk but it actually broke ROS install and I had to intall it again).

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: Kinetic
  - Autoware branch: Master

FIX : 
I managed to fix the issue with the missing libs (openni2 and vtkRenderingPythonTkWidgets) in ubuntu 16.04

- For openni2 :
---> sudo apt-get install the following : openni2-doc && openni2-utils && openni-doc  && openni-utils
---> sudo apt-get install the following :
libopenni0                        libopenni-sensor-pointclouds0
libopenni2-0                      libopenni-sensor-pointclouds-dev
libopenni2-dev                 
libopenni-dev           

- For the issue with vtkRenderingPythonTkWidgets
---> I created a symbolic link to the existing vtkRenderingPythonTkWidgets lib (but which is in another folder)

sudo ln -s /usr/lib/python2.7/dist-packages/vtk/libvtkRenderingPythonTkWidgets.x86_64-linux-gnu.so /usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so

I hope this can help those who experience similar issues.

amc-nu, 2018-03-06:
Thanks for sharing your solution @dlounis

closed 2018-03-06 by amc-nu

ningscapr, 2019-01-11:
Thank you so much for your solution! @dlounis