autowarefoundation/autoware #1076: Missing dependencies and "publish point" button causes rviz to crash for open planner

sumjos, 2018-02-12:
Hello developers, firstly thank you all for providing the Autoware software! 
I plan to integrate and test a custom path planner package in Autoware, and see it perform on Rviz.
I recently started working on Autoware and I am running into some issues related to missing package dependencies and following open planner tutorial .

## Issue 1: Missing dependecies
On running rosdep install --from-paths src --ignore-src -r -y , I get the following output:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
lattice_planner: Cannot locate rosdep definition for [libarmadillo-dev]
decision_maker: Cannot locate rosdep definition for [generated_messages]
cv_tracker: Cannot locate rosdep definition for [klt]
ndt_gpu: Cannot locate rosdep definition for [common]
as: Cannot locate rosdep definition for [module_comm_msgs]
ndt_cpu: Cannot locate rosdep definition for [common]
data_preprocessor: Cannot locate rosdep definition for [pcd_tutorial]
fusion: Cannot locate rosdep definition for [dpm]
catvehicle: Cannot locate rosdep definition for [obstaclestopper]
filters: Cannot locate rosdep definition for [octree]
registration: Cannot locate rosdep definition for [features]
velodyne_compression: Cannot locate rosdep definition for [compression_msgs]
viewers: Cannot locate rosdep definition for [cv_tracker_msgs]
kf: Cannot locate rosdep definition for [dpm]
Reading package lists...
Building dependency tree...
Reading state information...
Package libxerces-c2-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source 

Autoware builds successfully though, and I am able to follow some tutorials, but I am not sure how important these dependencies are and if it is related to the next issue that I am facing. How should I fix this?

## Issue 2: "publish point" button causes rviz to crash for open planner
I am following this link: https://www.youtube.com/watch?v=FKM8v79X3_s ,for open planner tutorial.
In the last step, I try to add an obstacle in front of the moving vehicle and as I click on "Publish Point" on RViz, it suddenly crashes. Terminal logs are mentioned below (dataconnection.csv is not present in the sample data downloaded from Autoware home):

NODES
/
ff_waypoint_follower (ff_waypoint_follower/ff_waypoint_follower)

ROS_MASTER_URI=http://localhost:11311

[ERROR] [1518395807.464008395]: "map" passed to lookupTransform argument target_frame does not exist. 
Initialize Controller .. , 0,angle, velocity, 0, 100, 0.5
Current Steering Delay Factor = 0.2
[ INFO] [1518395808.464209843]: Origin : x=-14771.000000, y=-84757.000000, z=39.000000
ff_waypoint_follower initialized successfully

Can't Open Map File !, /home/AML/.autoware/data/map/vector_map/dataconnection.csvMap loaded from data with 73 lanes
X:-16687.357, Y:-86506.606, Z:48.638, A:0.96680603
Lon:35.13123811, Lat:136.5900087, Alt:48.638, Dir:0.96680603
[ INFO] [1518395815.134381310]: Received Start pose
[ INFO] [1518395818.734264727]: Received Goal Pose
Goal Found, LaneID: 2685, Distance : 0, Angle: 0

Info: PlannerH -> Plan (A) Path With Size (96), MultiPaths No(1) Extraction Time :
New DP Path -> 96
X:-16687.357, Y:-86506.606, Z:48.638, A:0.96680603
Lon:35.13123811, Lat:136.5900087, Alt:48.638, Dir:0.96680603
[ INFO] [1518395815.042619696]: init Simulation Rviz Pose Data: x=2.323055, y=-2.482887, z=0.000000, freq=0
Global Lanes Size = 1, Conv Size= 1, First Lane Size: 96
Performance Alert: Can't Find Lane Information in Global Path, Searching the Map :(
/home/AML/Autoware/ros/src/util/packages/runtime_manager/../../../.config/rviz/cmd.sh: line 44: 29179 Segmentation fault      (core dumped) rosrun rviz rviz

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: Kinetic-desktop-full
  - Autoware branch: Master
  - Installation instructions followed as mentioned on Autoware github page
  - Point cloud, vector map and all the data that I am using for tutorials is the "Sample Data" downloaded 
    from Autoware github page

hatem-darweesh, 2018-02-12:
Dear Sumit, 
the absence of dataconnection.csv will only affect the behavior of traffic light stop, but will not cause any crash, 
I think the problem is with rviz, you are missing some visualization plugins. kindly check rviz community, or reinstall rviz. also sometimes it is OpenGL problem. 
about the missing build packages, you need to install these packages as mentioned in the Autoware main page. some modules will not work without it. 

Best Regards,

sumjos, 2018-02-19:
Thank you @hatem-darweesh  
I tried reinstalling RViz but the problem still persists and could not find any know issue similar to this problem on rviz community. I will try out a few more things.

pntemi, 2018-02-24:
We ran into the same issue when trying to publish point in rviz. Here is the stack trace

`Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00000000400a2aae in ?? ()
(gdb) bt
#0  0x00000000400a2aae in ?? ()
#1  0x00007fffd58c53f9 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.390.25
#2  0x00007fffd58cc084 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.390.25
#3  0x00007fffd54900f5 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.390.25
#4  0x00007fffcb54e9c5 in Ogre::GLRenderSystem::_render(Ogre::RenderOperation const&) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007ffff2b1d7f7 in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#6  0x00007ffff2b47fb9 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff2b48095 in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff2b068df in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff2b06757 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff2b1c474 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff2b58f51 in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff2afd77a in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff2afd5cb in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool)
    () from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff2afd11e in Ogre::RenderTarget::updateImpl() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff2afdac4 in Ogre::RenderTarget::update(bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff2bade5f in Ogre::RenderSystem::_updateAllRenderTargets(bool) ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff2cc44ba in Ogre::Root::_updateAllRenderTargets() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff2cc45a0 in Ogre::Root::renderOneFrame() ()
   from /usr/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff761c32a in rviz::VisualizationManager::onUpdate (this=0x14619c0)
    at /home/autoware/dev/src/rviz/src/rviz/visualization_manager.cpp:371
#20 0x00007ffff763e639 in rviz::VisualizationManager::qt_static_metacall (
    _o=0x14619c0, _c=QMetaObject::InvokeMetaMethod, _id=3, _a=0x7fffffffded0)
    at /home/autoware/dev/build/rviz/src/rviz/moc_visualization_manager.cpp:106
---Type <return> to continue, or q <return> to quit---
#21 0x00007ffff13acd2a in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff13b95c8 in QTimer::timerEvent(QTimerEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff13adbb3 in QObject::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff6d1205c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff6d17516 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff137e38b in QCoreApplication::notifyInternal(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff13d35ed in QTimerInfoList::activateTimers() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff13d3af1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff08d5197 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff08d53f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff08d549c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff13d47cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff137bb4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00007ffff1383bec in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00000000004011db in main (argc=1, argv=0x7fffffffe608)
    at /home/autoware/dev/src/rviz/src/rviz/main.cpp:42
` 
Any thoughts about what we could do to fix this?

yk-fujii, 2018-02-27:
@sumjos @pntemi 
These are problems in rviz and nvidia.
https://github.com/ros-visualization/rviz/issues/1082
https://github.com/ros-visualization/rviz/issues/1192

If you need to use "publish point" tool, please select one of the following:
- use OpFlag tool (you can select opflag at rviz menu's + button)
- use ROS Indigo

francisvac, 2018-02-27:
What is OpFlag and what is it used for?
This is the description I found:
Tool for sim_obj flags on the ground plane in rviz.

In the Open Planner, what is the method to simulate objects in the path using OpFlag?

quhezheng, 2018-03-21:
'OpFlag' solution is perfect!
It can replace the function of 'Publish point' on waypoint_clicker

quhezheng, 2018-03-21:
It crash only on Nvidia graphic card. Free of crash on regular computer