autowarefoundation/autoware #1079: Autoware and SICK LMS511

dlounis, 2018-02-13:
I need to use data from SiCK LMS511 LiDAR with Autoware to perform mapping and localization. I have noticed that the LMS511 is listed in the sensing tab of the runtime manager but as far as I understand, this LiDAR is not implemented. 

Please, did someone use this LiDAR with Autoware (or another device other than the once implemented in the RT manager) ?

Best regards.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:14.04
  - ROS Version:Indigo
  - Autoware branch:Master

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

kitsukawa, 2018-02-14:
The driver for SICK LMS511 is in develop branch now. It will be merged into master branch in next release.

dlounis, 2018-02-14:
Dear Yuki,
thanks for your reply. Since I don't want to jeopardize my current installation of autoware, is there a way of adding the driver without a check out to the develop branch ?

kitsukawa, 2018-02-14:
How about cherry-picking the commits related to SICK LMS511 driver?

dlounis, 2018-02-14:
Thanks Yuki, I'll try that.

closed 2018-02-14 by kitsukawa