autowarefoundation/autoware #1096: How to run catvehicle in Gazebo using Autoware's Astar obstacle avoid?

sumjos, 2018-02-19:
I have recently started using Autoware and I am unable to figure out how to run the catvehicle in Gazebo and control it through Autoware. I want to use the built in Astar package to detect obstacles in Gazebo world and be able to drive the vehicle autonomously. I am able to run the vehicle in RViz as per [this](https://www.youtube.com/watch?v=B3UUKFM6Hqg) tutorial.

Steps so far:
1. Start Autoware -> Gazebo
2. Open RViz with Gazebo.config

I see the world in RViz and Gazebo model seems fine and /point_raw topic is being published, but I could not find anything to move beyond this point.

Any help would be appreciated.
Thank you.

slivingston, 2018-03-13:
@sumjos did you find a solution?

hakuturu583, 2018-12-06:
Autoware does not support gazebo (catvehicle) since kinetic.
I am trying to integrate with LGSVL Simulator with Autoware.
#1724 
Please wait a bit!!

yukkysaito, 2019-02-16:
LGSVL simulator is integrated. Also, Old gazebo will be replace the new Gazebo
https://github.com/CPFL/Autoware/pull/1930

closed 2019-02-16 by yukkysaito