autowarefoundation/autoware #1101: Build error

kangdw91, 2018-02-21:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.  Sorry It doesn't work, Can you help this problem?
  1.
  1.

## Specifications

  - Ubuntu version: Ubuntu Gnome 16.04
  - ROS Version: Kinetic
  - Autoware branch: master
  - CUDA 9.1
  - QT 5.10.1
  - Opencv 3.4 

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/jacobian_workspace.cpp.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel.cpp.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel_factory.cpp.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel_impl.cpp.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/timeutil.cpp.o
[ 10%] Building C object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/os_specific.c.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/MapDrawer.cc.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/string_tools.cpp.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/property.cpp.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Optimizer.cc.o
[ 10%] Linking CXX static library /home/ros/Autoware/ros/devel/lib/libg2o.a
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/PnPsolver.cc.o
[ 10%] Built target g2o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Frame.cc.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/KeyFrameDatabase.cc.o
/home/ros/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc: In member function ‘void ORB_SLAM2::Map::extractVocabulary(ORB_SLAM2::ORBVocabulary*)’:
/home/ros/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Map.cc:423:50: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::vector<ORB_SLAM2::KeyFrame*>::size_type {aka long unsigned int}’ [-Wformat=]
  fprintf (stderr, "KF: %d\n", allKeyFrames.size());
                                                  ^
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Sim3Solver.cc.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Initializer.cc.o
[ 10%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_slam2.dir/src/Viewer.cc.o
[ 11%] Linking CXX static library /home/ros/Autoware/ros/devel/lib/liborb_slam2.a
[ 11%] Built target orb_slam2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

amc-nu, 2018-02-22:
There is no error contained in the log. Try compiling adding `-j1` or use docker

kangdw91, 2018-02-23:
Yes, I solve the problem 
Thank you

amc-nu, 2018-02-23:
great!

closed 2018-02-23 by amc-nu