autowarefoundation/autoware #1107: Integrate Chrono Vehicle Model

huckl3b3rry87, 2018-02-22:
# New Feature
Integrate Chrono (for simulating vehicle dynamics) into Autoware.

# Motivation
[Chrono](https://projectchrono.org/) can be very slow if too much fidelity is added (i.e. soft soil, etc.) there are some models that can be run in real time and the vehicle models are tested and suitable when an accurate representation of the vehicle is needed. For instance, Gazebo does not have tire models and powertrain dynamics etc. unless you develop them and I have not found any validated vehicle models in Gazebo. 

# Status
 We have built Chrono into ROS and integrated it into a Docker file that also builds Autoware. The branch is [here](https://github.com/huckl3b3rry87/Autoware/tree/feature/chrono).

tqshao, 2018-02-27:
# TODO 

- [ ] Create a Chrono package that take a trajectory as input and outputs the vehicle states.

- [ ] Add a "Chrono" button on Autoware GUI that launches Chrono when clicking on it.

- [ ] Include Chrono into the dockerfile.

huckl3b3rry87, 2018-02-27:
# TODO 2
- [ ] Add two Chrono buttons, one for trajectory following and one for path following
- [ ] Identify vehicle parameters for HMMWV 
- [ ] make the vehicle model run in real time

closed 2018-03-12 by huckl3b3rry87