autowarefoundation/autoware #1109: Integrate Moving Obstacle Detector

huckl3b3rry87, 2018-02-22:
# Feature
Integrate the ROS package [obstacle_detector](https://github.com/tysik/obstacle_detector) into Autoware.

# Motivation
The motication (and please let me know if I am wrong) is that (again from what I understand) the current obstacle detection algoritms in Autoware do not predict the, postion (x,y), velocty (x,y),  and size (assuming circular obstacles) .

tqshao, 2018-02-27:
# TODO

- [x] Install the obstacle_detector package as a submodule.

- [x] Make an obstacle_detector launch file so that it subscribes the point cloud topic from Autoware.

- [ ] Test the performance of the obstacle_detector package.

huckl3b3rry87, 2018-02-27:
#TODO 2
- [ ] Set up obstacle_detector as [per this](https://github.com/tysik/obstacle_detector)
- [ ] Set up and test obstacle_detector as an example in Gazebo (I  have some code for this), stationary platform
- [ ] Set up and test obstacle_detector as an example in Gazebo (I  have some code for this), moving platform
- [ ] set up obstacle detector_package with a real VLP16 (I have code for this)

closed 2018-03-12 by huckl3b3rry87