autowarefoundation/autoware #1110: Workflow Working with Autoware

huckl3b3rry87, 2018-02-22:
I have several projects that I am working on and I am managing a team of masters students (this semester) at the University of Michigan to get this done. Below I have a summary of all of the features that I will be adding.

# Projects / Features
- [ ]  Integrate [Gazebo](https://github.com/CPFL/Autoware/issues/758) into Autoware
We have done a large amount of this already but could use help finishing up adding the path planning and tracking algorithms and making sure that the dependencies that needed to be removed are OK. While eventually, we will use the Chrono vehicle model, Gazebo will still be used to simulate the sensors (LiDAR etc.) and the environment.

- [ ] [Chrono](https://github.com/CPFL/Autoware/issues/1107)
We want to add this so that we can use the validated and high fidelity vehicle models.

- [ ] [Moving Obstacle Avoidance](https://github.com/CPFL/Autoware/issues/1108)
This is the algorithm that we are interested in testing, first in simulation and then in experimentation.

- [ ] [obstacle_detector](https://github.com/CPFL/Autoware/issues/1108)
In order to test the moving obstacle avoidance algorithm, we need a package in Autoware that can estimate moving obstacle speed etc from raw LiDAR data.

# Workflow
I noticed the [workflow](https://github.com/CPFL/Autoware/blob/master/docs/en/branching_model.md) document for Autoware, but I I have may have a slightly different case since I will be developing multiple features and then integrating them into my master branch. I guess if all of my branches start from and Autoware master, then when the base implementation is done, I can then merge it into my master.  

@dejanpan how do you suggest that I structure my workflow so that I can make it easy for integration into Autoware?

slivingston, 2018-03-12:
did you find that the catvehicle model (in the sourcetree under ros/src/system/gazebo/catvehicle/) is not sufficient in terms of "Integrate Gazebo into Autoware" and "simulate lidar in Gazebo"?

huckl3b3rry87, 2018-03-12:
@slivingston the catvehicle model runs very slowly, it is not real time at all

huckl3b3rry87, 2018-03-12:
@slivingston @amc-nu given our projects timeline and our limited resources and moreover since we are still developing our software and everything should be tested on our end before we try to integrate it we decided to just develop our packages on our own and maybe when they are done and more stable it will make sense to try to add them to Autoware. If you are interested the repo is here 
https://github.com/JuliaMPC/MAVs

With that, I am closing all of our issues and we can reopen them in the future if we decide to.  Thanks

closed 2018-03-12 by huckl3b3rry87