autowarefoundation/autoware #1112: NDT mapping fails with GPU

likui01, 2018-02-26:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
ndt mapping doesnt work with gpu  option

Error: an illegal memory access was encountered /home/xxx/Autoware-master/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/NormalDistributionsTransform.cu 929
Error: all CUDA-capable devices are busy or unavailable /home/xxxx/Autoware-master/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/MatrixDevice.cu 21
[ndt_mapping-2] process has died [pid 78745, exit code 1, cmd /home/xxxxx/Autoware-master/ros/devel/lib/ndt_localizer/ndt_mapping __name:=ndt_mapping __log:=/home/xxxxx/.ros/log/3a7eb290-1a91-11e8-83fc-7085c25362be/ndt_mapping-2.log].
log file: /home/xxxx/.ros/log/3a7eb290-1a91-11e8-83fc-7085c25362be/ndt_mapping-2*.log

## Actual Behavior
(Processed/Input): (2319 / 4788)
Error: out of memory /home/xxxx/Autoware-master/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/SymmetricEigenSolver.cu 11
Error: invalid device pointer /home/xxxx/Autoware-master/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/MatrixDevice.cu 21
[ndt_mapping-2] process has died [pid 72120, exit code 1, cmd /home/xxxx/Autoware-master/ros/devel/lib/ndt_localizer/ndt_mapping __name:=ndt_mapping __log:=/home/xxxx/.ros/log/3a7eb290-1a91-11e8-83fc-7085c25362be/ndt_mapping-2.log].
log file: /home/xxxx/.ros/log/3a7eb290-1a91-11e8-83fc-7085c25362be/ndt_mapping-2*.log

## Steps to Reproduce the Problem

  1. start ndt mapping with GPU option 
  1.
  1.

## Specifications

  - Ubuntu version:16.04
  - ROS Version:kinetic
  - Autoware branch: fix/ndt mapping with gpu issue

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

anhnv3991, 2018-02-28:
@likui01 The error you described in "Bug" was solved in #1006.
The error in "Actual behavior" occurs when the size of data structures required for constructing the map exceed the GPU memory capacity. You can try approximate_ndt_mapping instead of ndt_mapping.

likui01, 2018-02-28:
@anhnv3991  i used the file #1006  but still i get the same error.
I have GeForce titanX, 12GB memory.
i am using the sample point cloud data of 7.6GB

anhnv3991, 2018-02-28:
@likui01 can you show me line 929 of file Autoware-master/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/NormalDistributionsTransform.cu?

likui01, 2018-02-28:
@anhnv3991  

	int full_size = valid_points_num;
	int half_size = (full_size - 1) / 2 + 1;

	while (full_size > 1) {
		block_x = (half_size > BLOCK_SIZE_X) ? BLOCK_SIZE_X : half_size;
		grid_x = (half_size - 1) / block_x + 1;

		grid.x = grid_x;
		grid.y = 1;
		grid.z = 6;
		matrixSum<<<grid, block_x>>>(gradients, full_size, half_size, 1, 6, valid_points_num);
		checkCudaErrors(cudaGetLastError());

		grid.y = 6;
		matrixSum<<<grid, block_x>>>(hessians, full_size, half_size, 6, 6, valid_points_num);
		checkCudaErrors(cudaGetLastError());

		sumScore<<<grid_x, block_x>>>(score, full_size, half_size);
		checkCudaErrors(cudaGetLastError());

		full_size = half_size;
		half_size = (full_size - 1) / 2 + 1;
	}

	checkCudaErrors(cudaDeviceSynchronize()); line 929

	MatrixDevice dgrad(1, 6, valid_points_num, gradients), dhess(6, 6, valid_points_num, hessians);
MatrixHost hgrad(1, 6), hhess(6, 6);

anhnv3991, 2018-03-02:
@likui01 strange. The code you showed me looks like it belongs to the old version of NormalDistributionTransform.cu. Are you sure that you have used the new file?
When fixing the error, I added some lines, so if any error occurs during the mapping process on GPU, it should be reported on line 962, instead of 929.
Which branch are you working on?

likui01, 2018-03-02:
master branch

amc-nu, 2018-03-02:
@likui01 as @anhnv3991 mentioned, this was fixed on #1006 which is was merged to `develop` branch.

Please checkout develop branch, test, and tell us your result.

likui01, 2018-03-08:
well, approximate ndt mapping works fine, so for now i am using that. 
the file that has computation of gradients into two steps has issue. the project does not compile with that file. 
i am working on master branch

closed 2018-03-10 by amc-nu