autowarefoundation/autoware #1115: Cannot launch node of type [cv_tracker/ssd_unc]

trooperli, 2018-02-27:
Hello,
I follow the steps in https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/packages/cv_tracker/nodes/ssd/README.md to install caffe for ssd, but without success. The trace is shown below:

_... logging to /home/yli/.ros/log/faaf6e14-1bf0-11e8-9005-9cefd5fd7797/roslaunch-yli-Precision-5820-Tower-8664.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yli-Precision-5820-Tower:34075/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /ssd/gpu_device_id: 0
 * /ssd/image_raw_node: //image_raw
 * /ssd/network_definition_file: /home/yli/ssdcaff...
 * /ssd/pretrained_model_file: /home/yli/ssdcaff...
 * /ssd/score_threshold: 0.5
 * /ssd/use_gpu: True

NODES
  /
    ssd (cv_tracker/ssd_unc)

auto-starting new master
process[master]: started with pid [8689]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to faaf6e14-1bf0-11e8-9005-9cefd5fd7797
process[rosout-1]: started with pid [8717]
started core service [/rosout]
ERROR: cannot launch node of type [cv_tracker/ssd_unc]: can't locate node [ssd_unc] in package [cv_tracker]_

Following is my ssd.launch file:

_<launch>
	<arg name="car" default="true"/>
	<arg name="pedestrian" default="false"/>
	<arg name="use_gpu" default="true"/>
	<arg name="gpu_device_id" default="0"/>
	<arg name="score_threshold" default="0.5"/>
	
	
	
	<arg name="network_definition_file" default="$(env HOME)/ssdcaffe/models/VGGNet/VOC0712/SSD_512x512/deploy.prototxt"/>
	<arg name="pretrained_model_file" default="$(env HOME)/ssdcaffe/models/VGGNet/VOC0712/SSD_512x512/VGG_VOC0712_SSD_512x512_iter_120000.caffemodel"/> 

	<arg name="camera_id" default="/" />
	<arg name="image_src" default="/image_raw"/>
	
	
	<node pkg="cv_tracker" name="ssd" type="ssd_unc">
		<param name="network_definition_file" type="str" value="$(arg network_definition_file)"/>
		<param name="pretrained_model_file" type="str" value="$(arg pretrained_model_file)"/>
		<param name="use_gpu" type="bool" value="$(arg use_gpu)"/>
		<param name="score_threshold" type="double" value="$(arg score_threshold)"/>
		<param name="gpu_device_id" type="int" value="$(arg gpu_device_id)"/>
		<param name="image_raw_node" type="str" value="$(arg camera_id)$(arg image_src)"/>
	</node>
	
</launch>_

Would someone please help me to take a look? 

Thanks,
Greg

amc-nu, 2018-03-02:
@trooperli according to your error log:
`ERROR: cannot launch node of type [cv_tracker/ssd_unc]: can't locate node [ssd_unc] in package [cv_tracker]`

The node was not compiled. 

make sure you cloned and compiled ssd in your home directory as indicated in the README, otherwise you'll need to change the `CMakeLists.txt`.

Then try compiling again with `./catkin_make_release`.

amc-nu, 2018-03-06:
@trooperli Did you fix this?

trooperli, 2018-03-06:
Hi Abraham,
Thanks for your help!! I think the reason why I failed the first time is
because I didn't  compiling again with ./catkin_make_release. Maybe it is a
good idea to add it to README?

Best,
Greg

amc-nu, 2018-03-06:
glad to hear you fixed it

closed 2018-03-06 by amc-nu