autowarefoundation/autoware #1119: ndt_matching of pcl_anh_gpu has died

yk-fujii, 2018-03-01:
# Bug
When use 'pcl_anh_gpu', I got "operration not supported" error.
I think pthread_mutex can not use in the kernel function.

## Steps to Reproduce the Problem

  1. Use pcl_anh_gpu's ndt_maching on drivepx2

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: kinetic
  - Autoware branch: develop
  - Used Map: moriyama( only use bin_global*)
  
  - Machine: Nvidia DrivePX2
  - CPU: aarch64
  - HostMem: 8GiB
  - GPUMem: 4GiB

```
Error: operation not supported /home/nvidia/autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/Registration.cu 366
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ndt_matching-1] process has died [pid 13760, exit code -6, cmd /home/nvidia/autoware/ros/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/nvidia/.ros/log/adb9dbe8-1c6e-11e8-bf2f-4ccc6ad50d08/ndt_matching-1.log].
log file: /home/nvidia/.ros/log/adb9dbe8-1c6e-11e8-bf2f-4ccc6ad50d08/ndt_matching-1*.log

```

anhnv3991, 2018-03-02:
@yk-fujii No, this is not pthread's problem. Somehow the system on NVIDIA DrivePX2 does not support cudaHostRegister. Hence, the system returns an error right after cudaHostRegister is called.

yk-fujii, 2018-03-02:
@anhnv3991 
ah, I understand.

Is it difficult to implement without you using cudaHostRegister?
 Is it impossible with cudaHostAlloc?

yk-fujii, 2018-03-04:
@anhnv3991 
Great!
Could you create patch which is disable it on PX2(CPU arch is aarch64) only?

anhnv3991, 2018-03-04:
@yk-fujii OK.

amc-nu, 2018-03-18:
@yk-fujii @anhnv3991 can we close this?

kitsukawa, 2018-03-22:
This is fixed in #1139 .

closed 2018-03-22 by kitsukawa