autowarefoundation/autoware #1121: Setup the A* Path Planning Algorithm in Gazebo

huckl3b3rry87, 2018-03-01:
# Feature
Setup the [A* Path Planning Algorithm](https://github.com/CPFL/Autoware/tree/master/ros/src/computing/planning/motion/packages/astar_planner) (or others if there is a better one) in Gazebo. 

# TODO 
- [ ] 1) figure out the topics that this node subscribes and publishes to
- [ ] 2) establish/ensure connection to this node and its subscribed topics
- [ ] 3) test using system described in https://github.com/CPFL/Autoware/issues/1110

huckl3b3rry87, 2018-03-01:
@sumjos can you please take a look at these items?  Thanks!

sumjos, 2018-03-01:
Alright. I am going to start on these tasks.

huckl3b3rry87, 2018-03-07:
the branch is [here](https://github.com/huckl3b3rry87/Autoware/tree/feature/base_local_planner)

closed 2018-03-12 by huckl3b3rry87