autowarefoundation/autoware #1127: GPS coordinate system transform...

SyuanLiao, 2018-03-05:
I read the geo_pos_conv.cpp and fix2tfpose.cpp, I'm confused about these questions.....

1. Is the map frame same with the ECEF coordinate(which uses three-dimensional XYZ coordinates to describe the location of a GPS user or satellite)?
    map frame = ECEF coordinate?

2. How to generate the value(bold text) in tf.launch?

   !-- worldからmapへのtf --
   node pkg="tf"  type="static_transform_publisher" name="world_to_map" args="**14771 84757 -39 0 0 0** /world /map 10" 

Is releated to llh to ECEF transformation?

3. in geo_pos_conv.cpp, the function "geo_pos_conv::set_plane" uses 19 regions to transform for Japan. How to transform for Taiwan?

Thank you :)

jokla, 2018-04-12:
Hi @SyuanLiao Have you figured out some of your questions?  Thanks

kitsukawa, 2018-04-12:
@SyuanLiao 
1. Speaking of the sample map of Moriyama,  it is not ECEF coordinate sysytem. It is the Japanese rectangular coordinate system.
2. The value in tf.launch is determined according to the position where the map exist. For example, the map of Moriyama is located around (x,y,z)=(-14771, -84757, 39) in the Japanese coordinate system, so we publish transform (14771, 84757, -39) from world frame to map frame so that we can see the map in the center of world frame on RViz.
3. We don't know about the Taiwanese coordinate system. You need to implement the conversion from lat,lon to Taiwanese coordinate system. FYI, we are working on converting map into the universal coordinate system. If this is done, we can handle map regardless of the coordinate system of each country.

SyuanLiao, 2018-04-12:
@jokla Hi, you can reference [issue#492](https://github.com/CPFL/Autoware/issues/492).

@kitsukawa Hi, Thank you for your answers. :)
We referenced [issue#492](https://github.com/CPFL/Autoware/issues/492) and have done GPS coordinate for Taiwan.

now we have another question similar [issue#439](https://github.com/CPFL/Autoware/issues/439)
Sometimes the bug happened when it did NDT localization, and then we fixed it after followed your answer.
Our question is what caused it happened?

kitsukawa, 2018-04-16:
@SyuanLiao Glad to hear you can perform ndt_matching. I don't know exactly about the cause of that error, but the error does not occur after you build Autoware again by `./catkin_make_release`.

closed 2018-04-16 by kitsukawa

wsung1, 2018-05-23:
Hi, @SyuanLiao, I think I ran into the almost same problem you had before (regarding the GPS coordinate out of Japan). If you don't mind, would you elaborate more about how you could solve this problem? Thank you!

SyuanLiao, 2018-05-25:
Hi, @wsung1 , you can reference [issue#492](https://github.com/CPFL/Autoware/issues/492).
Use the functions `LocalCartesian (real lat0, real lon0, real h0=0, const Geocentric &earth=Geocentric::WGS84())` and `void Forward (real lat, real lon, real h, real &X, real &Y, real &Z)
` to set the gps of the origin of the map and get the local gps position.
library Document: [here](https://geographiclib.sourceforge.io/html/classGeographicLib_1_1LocalCartesian.html)

javcursor, 2018-09-17:
Hi, I've done the suggested changes, the conversion has coherent values of X,Y,Z after check the sign of latitude and longitude. I've added a static transform between base_link and gps because they was separated.
I want know why the baselink pose is orbiting randomly over the map.
I let a video with the actual behavior, the dataset and my src folder with modifications (https://www.dropbox.com/sh/2mv422ww4y2rh1b/AABagU-tDogSLqoATnGbU7t4a?dl=0):

- geo_pos_conv.cpp
- nmea2tfpose_core.cpp
- ndt_matching.launch

I'm using Ubuntu 16.04, ROS Kinetic and Autoware 1.8, any help is welcome.

HHF111, 2020-03-08:
@SyuanLiao  Thanks for all the info. May I ask you how to import GeographicLib to autoware?  Thx!