autowarefoundation/autoware #1132: Bug report in NDT localization

wangweiw4y4, 2018-03-05:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

## Bug report
A small bug recently was found in the ndt localization code that the calculation of the angular velocity didn't consider the discontinuity of the yaw angle, then the angular velocity will jump when the robot is operated near the yaw angle of +/- 180 degree. This causes the instability of some kinds of yaw-rate dependent controllers.
  
And the problem can be fixed it by the following codes:
    diff_yaw = current_pose.yaw - previous_pose.yaw;
    if(diff_yaw > M_PI)
    diff_yaw = diff_yaw - 2* M_PI;
    if(diff_yaw < -M_PI)
    diff_yaw = diff_yaw + 2* M_PI;
    diff = sqrt(diff_x * diff_x + diff_y * diff_y + diff_z * diff_z);

## Specifications:

  - Ubuntu version: 16.04
  - ROS Version: Kinect
  - Autoware branch: master

kitsukawa, 2018-03-06:
@wangweiw4y4 Thank you for the report. Do you have any idea on the solutions for this?

wangweiw4y4, 2018-03-06:
@kitsukawa I think we can just fix it by the following codes:
diff_yaw = current_pose.yaw - previous_pose.yaw;
if(diff_yaw > M_PI)
diff_yaw = diff_yaw - 2* M_PI;
if(diff_yaw < -M_PI)
diff_yaw = diff_yaw + 2* M_PI;
diff = sqrt(diff_x * diff_x + diff_y * diff_y + diff_z * diff_z);

kitsukawa, 2018-03-22:
@wangweiw4y4 Thank you for your comment. This is implemented in `calcDiffForRadian()` in cd928df6105b0afde4447958217fc14f047f29fc . This is in the develop branch and it will be merged into master in the next release.

closed 2018-03-22 by kitsukawa