autowarefoundation/autoware #1133: Vector Map Generation

Xi-Cai, 2018-03-05:
Hello Autoware team,

Thank you all for this awesome open-source platform. I already had a pretty good path-following demo by using Autoware. Basically I used **waypoint_loder** to import my waypoints and used **lane_planner** as the mission planer and **lattice palnner** as the motion planer. Also I used the pure-pursuit to follow the waypoint and in my real-time vehicle test, it works pretty well. Now I am planning to realize a basic "obstacle stop" function. (If there is another vehicles/pedestrian/bicycles in the lane, the ego vehicle will stop and when they are gone, the vehicle will resume the velocity). However, I have no idea how to simulate obstacle with these planners. In Rviz, the "simulated obstacle" is only for open planner. 

Any sugguestion will be appreciated. Thanks in advance.

Xi

hroachewilson, 2018-10-29:
@Xi-Cai did you manage to implement the obstacle stop demo?

hatem-darweesh, 2019-03-24:
@Xi-Cai you can simulate with lgsvl, gazebo or carla for simulation with Autoware, it will take additional work to understand how these simulator works and how to integrate with autoware, but finally it is much better than any simulation inside Rviz.

closed 2019-03-24 by hatem-darweesh