autowarefoundation/autoware #1147: Build error due to multi lidar pull request

huckl3b3rry87, 2018-03-10:
Using the develop branch I get this issue, due to a missing dependancy
```
[ 85%] Building CXX object computing/planning/motion/packages/astar_planner/CMakeFiles/velocity_set.dir/nodes/velocity_set/velocity_set.cpp.o
/usr/bin/ld: cannot find -lQt5::Core
collect2: error: ld returned 1 exit status
sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeFiles/autoware_camera_lidar_calibration_node.dir/build.make:824: recipe for target '/home/febbo/Documents/workspace/Autoware/ros/devel/lib/autoware_camera_lidar_calibrator/autoware_camera_lidar_calibration_node' failed
make[2]: *** [/home/febbo/Documents/workspace/Autoware/ros/devel/lib/autoware_camera_lidar_calibrator/autoware_camera_lidar_calibration_node] Error 1
CMakeFiles/Makefile2:20190: recipe for target 'sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeFiles/autoware_camera_lidar_calibration_node.dir/all' failed
make[1]: *** [sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeFiles/autoware_camera_lidar_calibration_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Building CXX object computing/planning/motion/packages/astar_planner/CMakeFiles/velocity_set.dir/nodes/velocity_set/libvelocity_set.cpp.o
[ 85%] Linking CXX executable /home/febbo/Documents/workspace/Autoware/ros/devel/lib/object_map/laserscan2costmap
[ 85%] Built target laserscan2costmap
/home/febbo/Documents/workspace/Autoware/ros/src/computing/perception/semantics/packages/object_map/nodes/points2costmap/points2costmap.cpp:115:18: warning: ‘std::vector<int> {anonymous}::filterCostMap(std::vector<int>&)’ defined but not used [-Wunused-function]
 std::vector<int> filterCostMap(std::vector<int>& cost_map)
                  ^
[ 85%] Linking CXX executable /home/febbo/Documents/workspace/Autoware/ros/devel/lib/object_map/points2costmap
[ 85%] Linking CXX shared library /home/febbo/Documents/workspace/Autoware/ros/devel/lib/libobject_map_utils_lib.so
[ 85%] Built target points2costmap
[ 85%] Built target object_map_utils_lib
[ 85%] Linking CXX executable /home/febbo/Documents/workspace/Autoware/ros/devel/lib/object_map/potential_field
[ 85%] Built target potential_field
[ 85%] Linking CXX executable /home/febbo/Documents/workspace/Autoware/ros/devel/lib/multi_lidar_calibrator/multi_lidar_calibrator
[ 85%] Linking CXX executable /home/febbo/Documents/workspace/Autoware/ros/devel/lib/astar_planner/velocity_set
[ 85%] Built target multi_lidar_calibrator
[ 85%] Built target velocity_set
[ 85%] Linking CXX shared library /home/febbo/Documents/workspace/Autoware/ros/devel/lib/libroad_occupancy_processor_lib.so
[ 85%] Built target road_occupancy_processor_lib
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
It seems that [this pull request](https://github.com/CPFL/Autoware/commit/7438af84bf99174a5185d81f2f73e6ea275f67fa) caused this.

@kitsukawa any idea what is failing here?

amc-nu, 2018-03-10:
Please add qt5core to the required packages.

huckl3b3rry87, 2018-03-10:
@amc-nu I added 
```
cmake_minimum_required(VERSION 2.8.12)
project(multi_lidar_calibrator)

find_package(catkin REQUIRED COMPONENTS
        roscpp
        std_msgs
        sensor_msgs
        geometry_msgs
        pcl_ros
        pcl_conversions
        qt5core
        )
```
But it still has errors
```
Parallel;vtkRenderingFreeTypeFontConfig;vtkIOFFMPEG;vtkIOMPIImage;vtkIOGDAL (Required is at least version "1.7") 
-- +++ processing catkin package: 'kitti_launch'
-- ==> add_subdirectory(util/packages/kitti_pkg/kitti_launch)
-- +++ processing catkin package: 'multi_lidar_calibrator'
-- ==> add_subdirectory(sensing/fusion/packages/multi_lidar_calibrator)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "qt5core" with any
  of the following names:

    qt5coreConfig.cmake
    qt5core-config.cmake

  Add the installation prefix of "qt5core" to CMAKE_PREFIX_PATH or set
  "qt5core_DIR" to a directory containing one of the above files.  If
  "qt5core" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  sensing/fusion/packages/multi_lidar_calibrator/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'qt5core'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "qt5core" with any
  of the following names:

    qt5coreConfig.cmake
    qt5core-config.cmake

  Add the installation prefix of "qt5core" to CMAKE_PREFIX_PATH or set
  "qt5core_DIR" to a directory containing one of the above files.  If
  "qt5core" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  sensing/fusion/packages/multi_lidar_calibrator/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/febbo/Documents/workspace/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/febbo/Documents/workspace/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

huckl3b3rry87, 2018-03-10:
Also, these are some missing dependancies
```
sudo apt install ros-kinetic-jsk-recognition-msgs
 sudo apt install ros-kinetic-grid-map-ros
```

amc-nu, 2018-03-10:
Yes, those must be already in the README for the develop branch.

huckl3b3rry87, 2018-03-10:
I am assuming that I need to be building with qt5 not
```
$ qmake --version
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu
```

huckl3b3rry87, 2018-03-10:
@kitsukawa can you please fix your multi lidar calibration package?

amc-nu, 2018-03-10:
@huckl3b3rry87 I created this package. But I think this package is not the problem.
What’s your Ubuntu version?

amc-nu, 2018-03-10:
Found the problem I’ll push a fix.

amc-nu, 2018-03-11:
@huckl3b3rry87 can you please test #1149 and check if your error is fixed?

slivingston, 2018-03-11:
@amc-nu I was able to reproduce the error in the opening post, but instead of #1149, I find that it is sufficient to
```diff
diff --git a/ros/src/sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeLists.txt b/ros/src/sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeLists.txt
index 5acebbf2..5ad9a5d5 100644
--- a/ros/src/sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeLists.txt
+++ b/ros/src/sensing/fusion/packages/autoware_camera_lidar_calibrator/CMakeLists.txt
@@ -17,6 +17,7 @@ find_package(catkin REQUIRED COMPONENTS
 
 find_package(OpenCV REQUIRED)
 find_package(PCL)
+find_package(Qt5Core REQUIRED)
 
 catkin_package(CATKIN_DEPENDS
     cv_bridge
```

slivingston, 2018-03-11:
so, same as your changeset at https://github.com/CPFL/Autoware/pull/1149/commits/907e3a5562e1715e4d6712e5dfde317ad19f1ef4 except no need to add `${Qt5Core_LIBRARIES}` line in `target_link_libraries()`

amc-nu, 2018-03-11:
@huckl3b3rry87 can you check if this PR and the solution proposed by @slivingston fixes your compilation error. I haven't been  able to reproduce this error in any of my setups.

amc-nu, 2018-03-11:
@slivingston thanks for testing.

amc-nu, 2018-03-12:
fixed in #1149

closed 2018-03-12 by amc-nu