autowarefoundation/autoware #1150: issuse about libymc_can_2.0.a

mylxiaoyi, 2018-03-11:
when i compiled the master branch of Autoware with gcc6 on Debian9, i got the following errors:

```
/usr/bin/ld: actuation/vehicles/packages/ymc/lib/libymc_can_2.0.a(cansend.cpp.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld:actuation/vehicles/packages/ymc/lib/libymc_can_2.0.a(libcansend.cpp.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: ctuation/vehicles/packages/ymc/lib/libymc_can_2.0.a(ymc_can.cpp.o): relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
```

## Specifications

  - OS version: Debian 9 Stretch, gcc 6
  - ROS Version: lunar
  - Autoware branch: master

Thanks.

amc-nu, 2018-03-11:
@mylxiaoyi 
this is a proprietary library, so we don't have the source code available to recompile it for newer compilers.
If you don't require this package I recommend you to delete the `ymc` folder inside 
`Autoware/ros/src/actuation/vehicles/packages/` so you can continue with compilation.

mylxiaoyi, 2018-03-12:
@amc-nu 
Thanks.

closed 2018-03-12 by mylxiaoyi