autowarefoundation/autoware #1152: ndt_matching  process died ~

zhuxuekui3, 2018-03-13:
Fill-out only one section depending on whether you are reporting a bug or a new feature.
ndt.align 1100 col   process died

if run  "ndt.align(*output_cloud2, init_guess)"

# Bug
## Expected Behavior

## Actual Behavior

ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertion `lhs.cols()>0 && "you are using a non initialized matrix"' failed.
Aborted (core dumped)

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:  14.04
  - ROS Version:  indigo
  - Autoware branch: ndt_matching.cpp

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

zhuxuekui3, 2018-03-13:
i solved the problem:    catkin_make -DCMAKE_BUILD_TYPE=Release

jh0w3l05, 2018-08-09:
Hi,
I am also experiencing the same error but I build it with [catkin_make -DCMAKE_BUILD_TYPE=Debug]. Is there any way to make this work on a DEBUG build? I just want to understand the code. 
Thanks.

closed 2018-11-12 by yk-fujii