autowarefoundation/autoware #1157: Issue about opencv_apps/src/nodelet/lk_flow_nodelet.cpp

mylxiaoyi, 2018-03-14:
When I compiled master branch of Autoware, I got the following errors:

```
Autoware/ros/src/opencv_apps/src/nodelet/lk_flow_nodelet.cpp:162:92: error: call of overloaded ‘goodFeaturesToTrack(cv::Mat&, std::vector<cv::Point_<float> >&, int&, double, int, cv::Mat, int, int, double)’ is ambiguous
         cv::goodFeaturesToTrack(gray, points[1], MAX_COUNT, 0.01, 10, cv::Mat(), 3, 0, 0.04);
In file included from /usr/local/include/opencv2/imgproc/imgproc.hpp:48:0,
                 from /media/datadisk/source/ros_catkin_ws.lunar/install_isolated/include/cv_bridge/cv_bridge.h:44,
                 from src/opencv_apps/src/nodelet/lk_flow_nodelet.cpp:44:
/usr/local/include/opencv2/imgproc.hpp:1947:19: note: candidate: void cv::goodFeaturesToTrack(cv::InputArray, cv::OutputArray, int, double, double, cv::InputArray, int, bool, double)
/usr/local/include/opencv2/imgproc.hpp:1952:19: note: candidate: void cv::goodFeaturesToTrack(cv::InputArray, cv::OutputArray, int, double, double, cv::InputArray, int, int, bool, double)
```

## Specifications

  - OS version: Debian 9 Stretch
  - ROS Version: Lunar
  - Autoware branch: master
  - OpenCV: 3.4

I think it should be

`cv::goodFeaturesToTrack(gray, points[1], MAX_COUNT, 0.01, 10, cv::Mat(), 3, false, 0.04);`

amc-nu, 2018-03-14:
@mylxiaoyi I see you have another directory inside the `src` directory which doesn't belong to Autoware
`opencv_apps`. The package or packages inside that directory are causing this error.

mylxiaoyi, 2018-03-14:
Oh, I see, thanks.

closed 2018-03-14 by mylxiaoyi