autowarefoundation/autoware #1158: Understanding vector map format

alex-mocanu, 2018-03-15:
Hi,

Me and my colleagues working on developing an autonomous car are trying to decipher the meaning of the data fields in the csv vector map files (we are checking out the files in the quick start: area.csv, crosswalk.csv, etc.). We want to construct similar vector maps for the maps in the CARLA simulator.
We would be grateful if you could tell us what each data field in the vector map files represents.

aivo0, 2018-03-20:
Hi Alex. As I understand, the vector map format is not open source and nobody is willing to reveal it's exact specification, because HD mapping business is currenlty very competitive and has big business potential. 
I can give you and overview and rought descriptions, because we have tried to understand the format in our university. PM me in the Autoware Slack if you're interested. 
Aivo Olev

alex-mocanu, 2018-03-21:
Thank you a lot!
I left you a message on Slack.

closed 2018-04-09 by kitsukawa

DmitryRedin, 2018-12-10:
@alex-mocanu, @aivo0, Could you post the format here?

ruisir001, 2019-01-10:
@alex-mocanu , @aivo0 , @DmitryRedin , Could you post me the csv vector map specification, thank you !

DmitryRedin, 2019-01-14:
@ruisir001, [https://autoware.herokuapp.com/]

ruisir001, 2019-01-18:
@DmitryRedin , thank you!

hakunaMatataHub, 2019-06-17:
Here is the link to documentation of vector map ( I don't know why no one shared this yet ?!) 
Please visit this site [https://tools.tier4.jp/vector_map_builder/user_guide/]
P.S - you may have to first create an account, login and access this.

hakunaMatataHub, 2019-06-17:
PPS - Signing up in this site is free as of now.

BigMouseFive, 2020-08-06:
> @alex-mocanu , @aivo0 , @DmitryRedin , Could you post me the csv vector map specification, thank you !

Could you post me the csv vector map specification, thank you !  790545771@qq.com

JWhitleyWork, 2020-08-06:
@BigMouseFive Unfortunately, the format is proprietary. This was much of the reason that we started using Lanelet2 as of 1.13.0. If you create an account at https://tools.tier4.jp, you'll see that their map tool allows for the creation of either a Vector Map for versions prior to 1.13.0 or a Lanelet2 map for versions 1.13.0 and above. Their User Guide at https://tools.tier4.jp/vector_map_builder/user_guide/ does have _some_ information on the map format but the CSV file contents and header descriptions are not available.

aivo0, 2020-08-06:
Go with the Lanelet2 as suggested. 
We never completely figured out the format. I have been off the project for a while but now this startup is supplying the maps four our self-driving car: https://eyevi.tech/ . I am sure you can ask them for some tips.