autowarefoundation/autoware #1168: Bug report in gmsl camera

vehicularkech, 2018-03-19:
Can not launch gmsl camera node.
# Bug
## Expected Behavior
when press Sekonix 3322/3323 GMSLCamera ,the gmsl camera node would start.

## Actual Behavior
when click gmsl camera button,error happens:
" Invalid roslaunch XML syntax: not well-formed (invalid token): line 1, column 0
The traceback for the exception was written to the log file"

## solutions:
in "...src/driveworks/packages/autoware_driveworks_interface/launch/gmsl_camera_interface.launch" file
change 
"run  : roslaunch autoware_driveworks_interface gmsl_interface" 
to
"run  : roslaunch autoware_driveworks_interface gmsl_interface.launch"

## Specifications

  - Ubuntu version: 16.04
  - ROS Version: ros kinect
  - Autoware branch: master

amc-nu, 2018-03-22:
this node only works on the DrivePX2

vehicularkech, 2018-03-22:
yes, i run Autoware  on the DrivePX2 and error happens:
" Invalid roslaunch XML syntax: not well-formed (invalid token): line 1, column 0
The traceback for the exception was written to the log file".

amc-nu, 2018-03-22:
can you try pulling back again the docker image?

closed 2018-04-07 by amc-nu

4n00815, 2019-03-19:
I had similar problem, solved by changing the line 64 in the file ~/Autoware/ros/src/util/packages/runtime_manager/scripts/sensing.yaml from
`run : roslaunch autoware_driveworks_interface gmsl_interface`
to
`run : roslaunch autoware_driveworks_interface gmsl_interface.launch`