autowarefoundation/autoware #1171: NDT_matching uses GPU as default

manato, 2018-03-19:
# Bug
## Expected Behavior
`ndt_matching` executed from runtime-manager with `pcl_generic` method type option is expected to run on CPU. 

## Actual Behavior
It runs on GPU.
(I confirmed it by `nvidia-smi -l 1` while  `ndt_matching` is running. The command output shows a process named `ndt_matching` consumes GPU resource)

## Steps to Reproduce the Problem

  1. Execute `ndt_matching` with `pcl_generic` option from runtime-manager

## Specifications

  - Ubuntu version: 14.04
  - ROS Version: Indigo
  - Autoware branch: develop

YamatoAndo, 2018-03-27:
Thank you for the report.
The GPU resource  is consumed at the time of instance creation of line 131.

https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp#L131

When using pcl_generic option, instance of the GPU implementation is generated, but the instance is NOT used and CPU implementation is used.
It is strange because it consumes useless GPU resources,
It works without problems.

In order to solve this problem, it is necessary to change the structure of the code,
I am going to fix it.

Nicholas-Mitchell, 2018-04-25:
**[Just a small idea - I don't even know for sure how AutoWare is detecting and allocating GPU resources]**

Would it be possible to make the resources _invisible_ to Autware by specifying environment variables i.e. without restructuring the code? This idea is stolen directly from Tensorflow, where I use a basic function like the following to decide which GPU devices are visible. If they are visible, Tensorflow reserves them all by default - even if they are not fully used.

```
def set_gpu(device=1):
    """Set which GPU devices are visible to Keras/Tensorflow using PCI_BUS_ID

    An alternative for direct use within a jupyter notebook, using magics:
    %env CUDA_DEVICE_ORDER=PCI_BUS_ID
    %env CUDA_VISIBLE_DEVICES=1

    Parameters
    ----------
    device : comma-separated list of device IDs e.g. "1", or "0,2", optional

    Returns
    -------
    out : None. prints success message

    """
    try:
        os.environ["CUDA_DEVICE_ORDER"] = "PCI_BUS_ID"   # see issue #152
        os.environ["CUDA_VISIBLE_DEVICES"] = device
        print('Successfully set visibility to {}'.format(device))
    except Exception as e:
        print('Failed to set GPU device visibility!')
        print(e)
```

Nicholas-Mitchell, 2018-05-11:
> I confirmed this is fixed in #1248
> @Nicholas-Mitchell <https://github.com/Nicholas-Mitchell> Can you make a
> new issue regarding the GPU resource allocation?

I can make a new issue if you would like yo pursue the possibility;
however, It is actually not an issue for me personally.

Shall I create one?