autowarefoundation/autoware #1175: question about the calibration of 2D lidar with respect to the camera

ttomchy, 2018-03-21:
Hi , I have noticed that in the autoware package contains the calibration of camera and 2D lidar. However , I could not find the algorithm about it .Is there are any materials about it? Or any wikis about it?

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS Version:
  - Autoware branch:

# New Feature
Describe where does new feature request come from. Add high-level picture of how the new feature should work.

amc-nu, 2018-03-22:
@ttomchy the calibration toolkit is based on
```
@InProceedings{bib:calibration_toolkit_base,
  author    = {He, Mengwen and Zhao, Huijing and Cui, Jinshi and Zha, Hongbin},
  title     = {{Calibration method for multiple 2d LIDARs system}},
  booktitle = {Proc. of the IEEE International Conference on Robotics and Automation},
  year      = {2014},
  pages     = {3034--3041}
}
```

closed 2018-03-22 by amc-nu

ttomchy, 2018-08-02:
@amc-nu Thank you for providing the calibration of camera and the velodyne. I wonder the algorithm is used for calibrating the camera and the Velodyne ? Can you give the name of the paper you used ? I want to have read the paper.