autowarefoundation/autoware #1197: ndt_mapping : how to use odometry?

dlounis, 2018-04-04:
I want to use odometry from the wheels with approximate_ndt_mapping. Please, how can I do that ? Are there any naming rules (frame_id) for the odometry messages ?

Thanks a lot for your help.

YamatoAndo, 2018-04-05:
@dlounis 
First, publish odometry msg as "/ odom_pose".
Next, please check "Use Odometry" from "app" tab of approximate_ndt_mapping.

There is no definite naming rule for frame_id of odometry msg.
However, approximate_ndt_mapping is created on the premise that odometry msg is the trajectory of base_link in the odom coordinate system.
So please make sure frame_id = "odom", child_frame_id = "base_link"

dlounis, 2018-04-05:
Thanks.
My odometry messages have frame_id = "odom" but the child_frame_id is empty. Please, is there a way to rewrite the child_frame_id in a new bag file ?

Regards

YamatoAndo, 2018-04-05:
@dlounis 
approximate_ndt_mapping does not check the name of child_frame_id.
So it does not matter even if it is empty.

dlounis, 2018-04-05:
thanks.
Is it normal that I still have mapping errors (mismatched scans on a straight portion of the road) even with odometry ?

YamatoAndo, 2018-04-06:
@dlounis
I can not judge that I do not see rosbag, but there is a possibility that a mismatch will occur.
Would you please show me the map and rosbag if you'd like it?

dlounis, 2018-04-06:
I think there is a mismatch. I have 9864 messages for points_raw topic and 19664 messages for /odom_pose topic. I think that both topics should have the same number of messages. Am I right ? I will check if I can share the data.

Regards.

YamatoAndo, 2018-04-06:
@dlounis 
There is no need to match the number of messages in points_raw topic and odom_pose topic.
So the cause is different.

dlounis, 2018-04-16:
Dear @YamatoAndo,
unfortunately, I am still stuck. However I have noticed (after several attempts to make it work) that the issue with odometry occurs mainly after passing roundabouts. It seems that NDT is more sensitive to errors on the angular parameters of the transforms (?)
Any suggestion is welcome.

Regards.

yk-fujii, 2018-05-14:
I think "how to use" topic is ended. 
Please divide issue thread.
Thanks.

kfunaoka, 2018-07-13:
Please divide issue thread.

closed 2018-07-13 by kfunaoka