autowarefoundation/autoware #1199: NDP mapping use no vehicle ego motion signal?

Moon-sik, 2018-04-04:
I'm beginner of ROS and Autoware.
 I want to mapping the road environment by using Autoware mapping tool.

But I have a question...
 To make the map, I don't neet to measure and save the vehicle motion signal like velocity, yaw rate and etc.,
 The NDT mapping tech. don't need other signals except point cloud data of lidar?

Thanks a alot..^^

YamatoAndo, 2018-04-05:
@Moon-sik 
You can create a map with only point cloud data of lidar.
Please refer to the video below. 
https://www.youtube.com/watch?v=ss6Blrz23h8

You can also use the vehicle motion signal together with the point cloud data of lidar to create a map.

kitsukawa, 2018-04-09:
@Moon-sik If you have further questions, please reopen this issue.

closed 2018-04-09 by kitsukawa