autowarefoundation/autoware #1203: Can't start RuntimeManager on lxterminal

andoh104, 2018-04-07:
# Bug
## Expected Behavior
Launch RuntimeManager on lxterminal 

## Actual Behavior
Fail to launch terminal

## Steps to Reproduce the Problem

  1../run
  

## Specifications

  - Ubuntu version:16.04
  - ROS Version:kinetic
  - Autoware branch:develop/master

## Cause
https://github.com/CPFL/Autoware/blob/master/ros/run
is wrong:
```
diff --git a/ros/run b/ros/run
index fe8349c..e2bcb9b 100755
--- a/ros/run
+++ b/ros/run
@@ -9,9 +9,6 @@ OPTION_RM_GEOMETRY='--geometry=50x10+500+0'
 OPTION_COMMAND='--command'
 
 
-MASTER_DISPLAY_OPTION="${OPTION_CORE_GEOMETRY} ${OPTION_TITLE}\=\"roscore\""
-RUNMGR_DISPLAY_OPTION="${OPTION_RM_GEOMETRY} ${OPTION_TITLE}\=\"runtime_manager\""
-
 if [ $(which gnome-terminal) ]; then
     TERMINAL=gnome-terminal
     GNOME_VERSION=$(gnome-terminal --version | cut -d '.' -f 2)
@@ -25,6 +22,8 @@ elif [ $(which xfce4-terminal) ]; then
     TERMINAL=xfce4-terminal
 elif [ $(which lxterminal) ]; then
     TERMINAL=lxterminal
+    OPTION_COMMAND='--command'
+    OPTION_TITLE='--title'
 elif [ $(which konsole) ]; then
     TERMINAL=konsole
     OPTION_WORKING_DIR='--workdir'
@@ -35,6 +34,9 @@ elif [ $(which konsole) ]; then
     OPTION_TITLE='-T'
 fi
 
+MASTER_DISPLAY_OPTION="${OPTION_CORE_GEOMETRY} ${OPTION_TITLE}=\"roscore\""
+RUNMGR_DISPLAY_OPTION="${OPTION_RM_GEOMETRY} ${OPTION_TITLE}=\"runtime_manager\""
+
 echo "Process Manager"
 
 gksudo --message "Please input password for launching process manager" \
(END)

```

yn-mrse, 2018-05-08:
I can not reproduce this problem. Is it a special environment?

andoh104, 2019-03-21:
Solved:
https://github.com/CPFL/Autoware/pull/1204

closed 2019-03-21 by andoh104